RobotLocomotion/drake #11753: Clean up official matlab support / recommendations

RussTedrake, 2019-06-29:
We occasionally get questions like #11742.  I think it's time to clean up our recommendations / official support for matlab.

Proposal:
 [ ] purge CallMatlab from the c++ (anyhow it is not sufficiently tested #6707); replace those examples with python
 [ ] add a simple example of calling pydrake from matlab using https://www.mathworks.com/help/matlab/call-python-libraries.html
 [ ] Update https://drake.mit.edu/matlab_bindings.html to recommend using the python bindings and point them to the example.
 [ ] Consider moving that example to a separate repo so that we can simplify the build files and [jenkins support matrix](https://drake-jenkins.csail.mit.edu/view/MATLAB/).

cc @jamiesnape

jamiesnape, 2019-07-01:
I think we had shared library issues with calling pydrake from MATLAB before. Did they resolve themselves with the latest MATLAB? Otherwise, yes, I am completely in favor.

RussTedrake, 2019-07-01:
i do not remember a shared library issue.  I remember that matlab didn't support 2d arrays in python.  (sigh).

Why do we still run a matlab job in pre-merge?  I would think that's safe to defer to nightly?

jamiesnape, 2019-07-01:
> i do not remember a shared library issue. I remember that matlab didn't support 2d arrays in python. (sigh).

We had issues with this test:
```
function mathematical_program_test		

  % Simple example of calling MathematicalProgram through the Python bindings.		

  % QP test		
 prog = py.pydrake.solvers.mathematicalprogram.MathematicalProgram();		
 x = prog.NewContinuousVariables(int32(2), 'x');		
 prog.AddLinearConstraint(x.item(int32(0)) >= 1.0);		
 prog.AddLinearConstraint(x.item(int32(1)) >= 1.0);		
 prog.AddQuadraticCost(py.numpy.eye(int32(2)), py.numpy.zeros(int32(2)), x);		
 result = prog.Solve()		

  % Note: int32(0) is kSolutionFound; can't reference it directly.		
 assert(result == py.pydrake.solvers.mathematicalprogram.SolutionResult(int32(0)));		

  x_sol = prog.GetSolution(x);		
 assert(abs(x_sol.item(int32(0)) - 1.0) < 1e-6);		
 assert(abs(x_sol.item(int32(1)) - 1.0) < 1e-6);
```
See also https://github.com/RobotLocomotion/drake/issues/7482.

> Why do we still run a matlab job in pre-merge? I would think that's safe to defer to nightly?

Just because we always have.

jamiesnape, 2019-07-01:
Maybe we should start with purging mention of `last_sha_with_original_matlab` from the docs (the tag can stick around)? We can also remove or replace the MATLAB job from pre-merge.

RussTedrake, 2019-07-01:
I look back at that tag frequently, and others might, too.  But I agree that referencing it on the "Using drake from matlab" page leads people to believe they could build that branch, and I suspect that's not true. 

Ok, I agree on both.

jamiesnape, 2019-07-05:
MATLAB has been removed from pull request builds.

EricCousineau-TRI, 2019-08-07:
(Realized that I didn't post a public summary)
We did a quick [survey on Slack](https://drakedevelopers.slack.com/archives/C0JNB2E3S/p1564411395009600) on 2019-07-29. Screenshot of results (as of 2019-08-07):
> [image]

Takeaway is that neither `call_matlab` nor `pydrake` from within MATLAB are being actively used, and thus we will remove them.

I will update top-level to remove the checkbox for testing `pydrake` within MATLAB.

EricCousineau-TRI, 2019-08-08:
Remaining item is to merge `call_matlab.h` into `call_python.h` or something. Then this can be closed.

PR #11953 (2019-08-31): Collapse call_matlab into call_python
Closes #11753.

----
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11953)
closed 2019-08-31 by sherm1