RobotLocomotion/drake #11758: Use OS version of Eigen on non-Xenial

jwnimmer-tri, 2019-07-01:
We are using a pinned version of Eigen (3.3.3) because it used to be the case that the OS version of Eigen was too old on most platforms (Trusty and Xenial).  However, in Bionic we have 3.3.4 available.

So once we drop support for Xenial (#10606) there is no more reason to pin and download and install our own version of Eigen.

I'd like to start that transition now, conditioned on an `os != xenial` guard in `repository.bzl`, so that we can both de-risk it and and reap the benefits prior to the Xenail EOL date.

jwnimmer-tri, 2019-07-01:
Thoughts @EricCousineau-TRI and @jamiesnape?

\CC @soonho-tri

jamiesnape, 2019-07-01:
Good idea.

soonho-tri, 2019-07-01:
Nice!

EricCousineau-TRI, 2019-07-01:
Sounds good to me!

PR #12780 (2020-03-03): workspace: Use system Eigen, not downloaded bitbucket tag
Closes #11758.  Needs #12784 and #12785 first.

I decided not to try to do Ubuntu and macOS separately.  It seems clearest to purge the installed-cmake-config code all at once, instead of conditioning it by platform.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/12780)
closed 2020-03-03 by jwnimmer-tri