RobotLocomotion/drake #11779: Improve RotationMatrix and RigidTransform operator* to allow multiplication of R * S  where R and S are different types.

mitiguy, 2019-07-03:
Improve RotationMatrix operator* to allow multiplication of R * S  where R and S are different types of rotation matrices (e.g., R is double and S is AutoDiffXd).  This problem was seen in PR #11752 and a TODO(Mitiguy) was added there.

Background: The code below cannot use RotationMatrix operator*. 
Instead, it uses the underlying Eigen "magic" that allows multiplying different types.
Why? R_AB is of type <T> whereas R_BC is of type <double>, yet RotationMatrix operator* only works for R_AB * R_BC if the RotationMatrix types of R_AB and R_BC are the same.

  const Matrix3<T> m3 = R_AB().matrix() * R_BC.matrix();
  const RotationMatrix<T> R_AC(m3);

Similarly, for RigidTransform operator*.

sherm1, 2019-07-03:
Before doing this, consider why we need to mix the types. That seems tricky, and if it is unusual it may be better to cast the `double` object to a `T` explicitly, after which the T-flavored rotations can be multiplied cleanly.

Per f2f we're not sure that it would be better to hide this conversion since whoever is reading the code _should_ be worried about the odd derivative that will result from these mixed types.

SeanCurtis-TRI, 2019-07-08:
FTR What I'd *actually* suggested to @mitiguy was that we support multiplication between `RotationMatrix<T>` and `RotationMatrix<double>` (and similarly for `RigidTransform`).

The idea being that it's valid for the scalars themselves and the Eigen types. So, it would make sense if we could do the same thing with our abstractions.

In this case, I have poses that are "properties" (i.e., always double) that I need to multiply against poses of type `T`. It's nice if it just did the right thing without gymnastics at the call site.

mitiguy, 2019-07-08:
Issue #11779 has been closed in favor of issue #11785.

closed 2019-07-08 by mitiguy