RobotLocomotion/drake #11780: Add matplotlib-cpp as an external

hongkai-dai, 2019-07-06:
Currently in order to do simple visualization, I need to write code in python and use matplotlib. For C++ code, I need to add the python binding. It would be more convenient to visualize directly in C++.

[matplotlib-cpp](https://github.com/lava/matplotlib-cpp) is a header only library, such that we can call many matplotlib functions in C++ directly. The library uses MIT license. Can we add this library as an external in Drake?

jwnimmer-tri, 2019-07-08:
The difficulty is that it links against the python runtime -- it's not a standalone library.  We have been trying to avoid bringing in an embedded python interpreter within Drake, as it tends to cause challenges for build robustness, debugging experience, etc.

Could you elaborate on whether and how `drake/common/proto/call_python.h` (and so calling matplotlib directly) is suboptimal?  (See `drake/examples/cubic_polynomial/backward_reachability.cc` for an example.)

hongkai-dai, 2019-07-08:
Thanks Jeremy! I wasn't aware that we have `call_python`. I will try it.

closed 2019-07-08 by hongkai-dai