RobotLocomotion/drake #11781: buildcop: `snopt_solver_test`  has ASan issues

EricCousineau-TRI, 2019-07-08:
Can reproduce locally on latest `master` @ :
```
$ ( set -ex; CC=$(which clang); CXX=$(which clang++); bazel test -c dbg --config=snopt_fortran --config=asan --test_tag_filters=snopt //solvers:s
nopt_solver_test ; )
```

Failures:
https://drake-cdash.csail.mit.edu/testDetails.php?test=31002769&build=1193356 (xenial)
https://drake-cdash.csail.mit.edu/testDetails.php?test=31083270&build=1193482 (bionic)

EricCousineau-TRI, 2019-07-08:
@jamiesnape Looks like either your commit introduced a check that wasn't running before, or perhaps lost some filtering info...
Do you know which it might be?

\cc @tri-ltyyu

jamiesnape, 2019-07-08:
It would be the former (our previous config must have been fooling us that we were checking this). There is no filtering to apply. If this is a problem internal to SNOPT, since we cannot currently handle whitelists, I would be inclined to apply a patch to SNOPT/Fortran at build time to add attributes.

EricCousineau-TRI, 2019-07-08:
Aye. From briefly debugging, it looks like passing the filename is what causes the overrun:
https://github.com/RobotLocomotion/drake/blob/7b257f3247429a3771fb10647919a9b297511210/solvers/snopt_solver.cc#L783

If I disable file logging (indicated by the stack trace), the test fails (of course), but there no longer is a buffer overrun:
```patch
diff --git a/solvers/snopt_solver.cc b/solvers/snopt_solver.cc
index 7bcfdddab..25644653a 100644
--- a/solvers/snopt_solver.cc
+++ b/solvers/snopt_solver.cc
@@ -780,7 +780,8 @@ void SolveWithGivenOptions(
     print_file_name = print_file_it->second;
   }
   Snopt::sninit(
-      print_file_name.c_str(), print_file_name.length(), 0 /* no summary */,
+      // print_file_name.c_str(), print_file_name.length(), 0 /* no summary */,
+      nullptr, 0, 0,
       storage.iw(), storage.leniw(),
       storage.rw(), storage.lenrw());
   auto guard = MakeGuard([&storage]() {
```

jamiesnape, 2019-07-08:
Or maybe we can do similar to this for now?
https://github.com/RobotLocomotion/drake/blob/b2050ed81bd6e674da93406df8090e888988b12c/solvers/test/snopt_solver_test.cc#L118-L131

EricCousineau-TRI, 2019-07-08:
Ooh, nice! Yeah, trying that out.

jamiesnape, 2019-07-08:
Out of interest, can you try setting `strict_string_checks=0`?

EricCousineau-TRI, 2019-07-08:
Yup, patching `strict_string_checks=0` on current `master` (7b257f324) does suppress the error.

PR #11782 (2019-07-08): snopt: Completely disable test use of "Print file" under ASan
Resolves #11781

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11782)
closed 2019-07-08 by jamiesnape