RobotLocomotion/drake #11785: Improve RotationMatrix::cast() and RigidTransform::cast().

mitiguy, 2019-07-08:
Make the RotationMatrix::cast() method more robust.  It is currently limited by Eigen's cast() method for the matrix underlying this class.  Consider the following logic.
1. If relevant, use Eigen's underlying cast method.
2. Strip derivative data when casting from `<AutoDiffXd>` to `<double>`.
3. Call ExtractDoubleOrThrow() when casting from `<symbolic::Expression>`   to `<double>`.
4. The current RotationMatrix::cast() method incurs overhead due to its underlying call to a RotationMatrix constructor. Perhaps create specialized code to return a reference if casting to the same type,  e.g., casting from `<double>` to `<double>' should be very inexpensive.

Similarly for RigidTransform::cast().

jwnimmer-tri, 2019-07-08:
I am skeptical that teaching RotationMatrix / RigidTransform themselves how to discard data (items 2 and 3) are a good idea in the first place, or even if the idea is sound that they are the best tactical approach (as opposed to, e.g., overloading `ExtractDoubleOrThrow` for these types).  I would need to see example call site improvements before my skepticism is assuaged.

SeanCurtis-TRI, 2019-07-09:
Perhaps a bit more elaboration would be in order. We have use cases where we need to go from rotation/rigid transform templated on `T` to one templated on `double` -- typically happens at interfaces to external libraries (fcl, vtk, opengl, etc.).

What would be ideal is for the `cast()` operator to work consistently. Currently we can only cast double to autodiff or double to symbolic. If you ever need to go *to* double, you get localized exertions that get duplicated or get shared but live in parallel with the otherwise misleading `cast()` methods.

The core of this proposal is to have `cast<>()` more generally useful. Such that for a given `<T>` I can call `cast<double>()` and have:

1. Something that always compiles. (This is the baseline.)
2. Something that is cheap if `T = double`. (This is the second-pass optimization.)

The semantics would be a bit like get double or throw -- if the cast at runtime isn't valid, then it throws. Certainly, if I had a symbolic transform that consisted strictly of constants, that would be valid and one would get a double-valued transform.

What are you thoughts on this?

jwnimmer-tri, 2019-07-09:
My suspicion is that consistently using `ExtractDoubleOrThrow` as the idiom for downgrading from scalar-templated code to non-scalar-templated code is better for readability than having multiple different spellings.  I think the best way to evaluate this question is to illustrate both designs with callsite code on the table.

sherm1, 2019-07-09:
If I understand correctly, the issue here is the opposite -- T is AutoDiffScalar but the RotationMatrix expression is `RM<double>*RM<T>`. The result should be autodiffable, so `RM<double>.cast<T>()*RM<T>` would work. But ExtractDoubleOrThrow() wouldn't do the job.

jwnimmer-tri, 2019-07-09:
`RM<double>.cast<T>()*RM<T>` already works.

SeanCurtis-TRI, 2019-07-10:
@sherm1 We'd discussed two issues. The point you raise, and address by @jwnimmer-tri is one point. And, yes, it is simply resolved by casting away from double.

This issue touches the other direction -- "casting"  to double (although it suggests it's about casting to an arbitrary T).

As to Jeremy's point about different spellings, except for one thing, I'd say that `cast<Target>` should be the preferred spelling. However, We can't do `T foo; foo.cast<double>();` or some such thing.

So, maybe it would be better to overload `ExtractDoubleOrThrow()` to include larger aggregate structures instead of just single scalar values.

mitiguy, 2019-07-17:
Closing this issue for now.
We can reopen if it becomes relevant later.

closed 2019-07-17 by mitiguy