RobotLocomotion/drake #11800: mkdoc: Figure out why some of PiecewisePolynomial's docstrings are not getting generated

EricCousineau-TRI, 2019-07-16:
Encountered in this PR:
https://github.com/RobotLocomotion/drake/pull/11783

@m-chaturvedi Would you have time to look at this starting next week and possibly have a workaround / solution by next Wednesday (07/24)?

\cc @jamiesnape @mpetersen94

m-chaturvedi, 2019-07-25:
They are not getting generated because we have [`@exclude_from_pydrake_mkdoc`](https://github.com/RobotLocomotion/drake/blob/004864a1ef09583754b8573f287fd2658d18aab3/third_party/com_github_pybind_pybind11/mkdoc.py#L774) for the both of them [1](https://github.com/RobotLocomotion/drake/blob/004864a1ef09583754b8573f287fd2658d18aab3/common/trajectories/piecewise_polynomial.h#L163), [2](https://github.com/RobotLocomotion/drake/blob/004864a1ef09583754b8573f287fd2658d18aab3/common/trajectories/piecewise_polynomial.h#L173).

If we just remove the exclude, the problem doesn't get solved because both have the [same number of arguments, same argument names, as well as arg_types](https://github.com/RobotLocomotion/drake/blob/master/third_party/com_github_pybind_pybind11/mkdoc.py#L884-L893) (after sanitization   `conststdvector`, `conststdvector`).  One simple solution could be to just change the variable name. #11833

EricCousineau-TRI, 2019-07-25:
D'oh, that makes sense! Reviewed #11833, makes sense!

PR #11833 (2019-07-25): Fix docstring in trajectories_py.cc
Closes #11800

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11833)
closed 2019-07-25 by EricCousineau-TRI