RobotLocomotion/drake #11801: Validate a passed Context corresponds to the proper System

edrumwri, 2019-07-17:
I just spoke with a fellow Drake user who has been bit by a bug that I've been bitten by multiple times: one inadvertently passes the wrong Context to a system (usually happens when failing to pass in a subsystem Context).

Often, one will receive a puzzling error message. In this user's case, it was:

`abort: Failure at bazel-out/k8-opt/bin/systems/framework/_virtual_includes/context_base/drake/systems/framework/context_base.h:237 in MaybeGetFixedInputPortValue(): condition '0 <= index && index < num_input_ports()' failed.
`

Since a System creates its Context, it can associate a unique identifier for that System with that Context. Or, if a unique identifier isn't desirable (I could imagine that we would want to be able to pass that Context to two identical systems), a hash might be the answer instead.

There might be an even easier solution but I think letting the user figure this one out on their own is no longer acceptable.

sherm1, 2019-07-17:
@drum are you able to run this is Debug? In theory you should get a much better message in that case. If that works, we can hopefully find a cheap subset of tests to leave enabled in Release.

sherm1, 2019-07-17:
Er, @edrumwri that is (see previous comment).

edrumwri, 2019-07-17:
That _was_ the output in debug mode. When I've experienced this problem in release mode before, I've been gifted with only a "Segmentation Fault" message :(

jwnimmer-tri, 2019-07-18:
To help inform a solution, please provide an acute example (real code or pseudo-code) that shows the user(s)' call sequence(s) that failed to produce a good diagnostic.

edrumwri, 2019-08-24:
Here's an example without _any_ warning or diagnostic that should produce one:

```
class FirstLeafSystem : public LeafSystem<double> {};
class SecondLeafSystem : public LeafSystem<double> {};

void ShouldBlow() {
  FirstLeafSystem sys1;
  SecondLeafSystem sys2;
  auto context = sys1.CreateDefaultContext();
  sys2.Publish(*context);
}
```

One possible response could be that an error might be encountered when that Context is used- though it might not be- and I can't think of a good reason for the particular use above not to be caught.

sherm1, 2019-08-24:
Yup, should have blown up.

PR #12551 (2020-01-14): Systems reject Contexts that they did not create
_Edit: Prior title was "Checks that the Context passed into System functions was created by ("belongs to") the System."_

Resolves #11801

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/12551)
closed 2020-01-14 by jwnimmer-tri