RobotLocomotion/drake #11802: pydrake: Make module organization more consistent (e.g. move `autodiffutils_py`, `symbolic_py`, etc. to `pydrake.common`)

EricCousineau-TRI, 2019-07-17:
From this comment:
https://github.com/robotlocomotion/drake/pull/11783#pullrequestreview-263001864

At present, we have things like:
```
pydrake.autodiffutils
pydrake.symbolic
```
Ideally, we should move them to reflect the module organization, either directly in `pydrake.common`, or underneath it as:
```
pydrake.common.autodiff
pydrake.common.symbolic
```

EricCousineau-TRI, 2020-10-26:
Updated to this address latest Drake Dev meeting notes.

FYI @jwnimmer-tri @ludwigschmidt-tri @RussTedrake

rpoyner-tri, 2026-11-13:
Closing as not planned. There has been no activity since 2020, and no one is working on reorganizing the module layout. The in-progress nanobind transition changes how bindings are organized, so a layout cleanup now would likely be redone. If there is a specific import-path inconsistency that bites you, please file a focused issue for it.