RobotLocomotion/drake #11807: MultibodyPlant API needs grouping & reordering for Doxygen clarity

sherm1, 2019-07-17:
MultibodyPlant has a large API. Many of the methods naturally fall into related groups (including overload sets with identical names and related methods with similar names). Some of the methods have been grouped nicely in the [Doxygen documentation](https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_multibody_plant.html) but many (> 100) are ungrouped, making it very difficult for users to find what they need.

While there may be many ways to re-engineer this API, that's out of scope for this issue. Instead, there are some short-term improvements that would make the existing API easier to navigate in Doxygen:
- create new `@name` groups so that all related methods are grouped together
- write some overall documentation for each group to explain the common theme therein
- reorder groups as necessary to put the most commonly-used methods near the top
- collect the many deprecated methods in there own group at the bottom.

For reviewer sanity, code _moving_ should be done in its own PR, where the description promises that _no_ code or existing-documentation changes have been made. That way the vast seas of green in Reviewable can be ignored. Then adding the new `@name` text and any required local method documentation changes can be put in a separate PR that does not do _any_ code moving. That could also be done in a multiple-PR series that does some moving, some updates, some more moving, etc.

Assigning to @mitiguy for consideration, with @amcastro-tri as likely feature reviewer.

amcastro-tri, 2019-07-18:
This would be a great improvement. Thanks @mitiguy and @sherm1.

SeanCurtis-TRI, 2019-07-18:
On PR management -- alternatively, you could do it in a 2-commit, single PR where one commit does the moving and the other commit adds content. It's simple to map those first two commits to two Reviewable revisions with subsequent changes adding single revisions. Feel free to ping me if you like this idea -- I've used it in the past with success.

amcastro-tri, 2019-07-18:
I like @SeanCurtis-TRI's proposal.

PR #12342 (2019-11-14): Reorder and regroup MultibodyPlant API
This PR reorders and regroups the MultibodyPlant API, with minimal changes to the actual documentation and no changes to the API.

Some fixes/additions to the documentation:
- added the reaction force output port to the picture at the top
- added a "table of contents" linking to each of the new groups 
- used `@anchor` consistently for every group

Reviewers -- please focus on whether the new groupings make sense, whether by regrouping I've made any of the text nonsensical, and please verify that the links work. Any substantive changes to the text should  [... cut]
closed 2019-11-14 by sherm1