RobotLocomotion/drake #11817: System with viz is cannot be automatically differentiated.

amcastro-tri, 2019-07-22:
## Background ##

We found this problem when attempting to simulate a Diagram that includes both the system's dynamics and visualization. There are multiple examples of this in master, for instance [the acrobot passive simulation.](https://github.com/RobotLocomotion/drake/blob/master/examples/multibody/pendulum/passive_simulation.cc).

That essentially defines the user facing API a user interacts with. In the example of the acrobot sim, a user can choose from a variety of integrators from the command line.

The dynamics for a continuous system given by `ẋ = f(t,x)` are defined as part of this Diagram by the one system that introduces the dynamics (say `MultibodyPlant` in the acrobot example).

Now, implicit integrators (`ImplicitIntegrator`), such as the `ImplicitEulerIntegrator`, compute during the solution process the gradient (aka Jacobian) of `f(t,x)`. The implicit integratros do this by using one of two methods: numerical differentiation or automatic differentiation ([through `AutoDiffXd` here](https://github.com/RobotLocomotion/drake/blob/5aac3cc19c8cbe60e3bbfbe5941ff953e56160d4/systems/analysis/implicit_integrator.h#L399)). A user can select which method to use with [`ImplicitIntegrator::set_jacobian_computation_scheme()`](https://github.com/RobotLocomotion/drake/blob/master/systems/analysis/implicit_integrator.h#L84)

 ## The problem ##
What happens now is that since our Diagram includes visualization (SG and all the LCM infrastructure to make that happen), the autodifferentiation process chokes (meaning it throws an exception stating there is at least a system that is not convertible).
How is it possible these integrators did work in the past you might ask? well, they work when they use "numerical" differentiation (the default), and thus no scalar conversion is needed.  They also work on toy problems (unit tests) with no viz.
Notice this happens under the hood, the user doesn't know about it. A user only want to run a sim under a given precision.

 ## How to reproduce the problem ##
In the acrobot example mentioned above, you can ask the `ImplicitEuler` integrator to use automatic differentiation to compute the system's dynamics' Jacobian with [`ImplicitIntegrator::set_jacobian_computation_scheme()`](https://github.com/RobotLocomotion/drake/blob/master/systems/analysis/implicit_integrator.h#L84). You will get an exception being thrown at runtime.

amcastro-tri, 2019-07-22:
Some ideas on how to solve this problem from a f2f with @jwnimmer-tri (posted in Slack `#dynamics`):
1. We could add transmog to the three items in question.
2. We could infer/label systems in a diagram as scalar-agnostic, and so don't need to transmog them, just reuse/clone the T=double object.
3. We could cull/remove systems that don't relate to the xcdot Jacobian (i.e., publish-only tentacles).
4. In any case, implicit integrator transmog should happen at setup time (as soon as we have a system + know we are in kAutomatic), not during evaluation -- we need to fail fast if the user made a mistake.

SeanCurtis-TRI, 2019-07-22:
Perhaps worth noting this applies to any `Diagram` with a `System` that has been explicitly declared as a child class of `LeafSystem<double>`:

  - DrakeVisualizer
  - ContactResultsToLcmSystem
  - FrameVisualizer
  - LcmSubscriberSystem
  - LcmLogPlaybackSystem
  - LcmPublisherSystem
  - LinearScalarSystem
  - PleidesSystem
  - SimpleContinuousTimeSystem
  - SimpleDiscreteTimeSystem
  - LcmInterfaceSystem
  - RgbdSensor (coming soon)
  - PoseBUndleToDrawMessage
  - RandomSource
  - SimpleMixedContinuousAndDsicreteTimeSystem
  - RobotStateDecoder
  - RobotCommandToDesiredEffortConverter
  - LcmImageArrayToImages
  - ImageWriter
  - RobotStateEncoder

Note: this lest isn't exhaustive. It was meant to exclude systems in `attic` or clearly used for testing purposes. (Apologies if something got mistakenly omitted or included.)

jwnimmer-tri, 2019-07-25:
Transcribing something else I've said, for the record:

Visualization should not be using `LcmPublisherSystem` + `PoseBundleToDrawMessage` + `DispatchLoadMessage` all distinctly.  Doing it with three separate pieces is both difficult to maintain (as shown by the choices above), as well as the wrong model for stateful RPCs in the systems framework.  `DRAKE_VIEWER_LOAD_ROBOT` is absolutely _not_ an initialization event; it is a resync of the state between the current geometry context and our in-memory model of the remote endpoint.  Our model of the remote endpoint is _not_ data in the context (an initialization bit), it is explicitly _outside_ of the context, because we need to resync the context against it.

Fundamentally, there should be a single `DrakeVisualizerSystem` that knows how to coordinate and interleave `DRAKE_VIEWER_LOAD_ROBOT` + `DRAKE_VIEWER_DRAW`, encapsulating all of the required stateful RPC details.  That new system could be typed on `<T>` (even symbolic) from the start.

Writing that system now is another good solution to the transmog question, instead of trying to add hacks to the current implementations.

EricCousineau-TRI, 2021-03-12:
@hongkai-dai Is this addressed with your PR #14569?

RussTedrake, 2021-07-31:
@hongkai-dai Is this addressed with your PR #14569?

hongkai-dai, 2021-07-31:
Yes I think it has been resolved. I have been using DrakeVisualize<AutoDiffXd> alongside simulating an AutoDiffXd Diagram.

closed 2021-07-31 by hongkai-dai