RobotLocomotion/drake #11822: Rename ContactSurface::grad_h_MN_M to grad_h_NM_M

DamrongGuoy, 2019-07-22:
Currently ContactSurface defines
     h_mn = e_m - e_n,
we want to rename it to h_nm (with the same definition as before) :
    h_nm = e_m - e_n,
which is more consistent with our notations ().

Therefore, grad_h_MN_M will rename to grad_h_NM_M with the same meaning as before (the vector points from N into M, and is expressed in frame M).

Apart from renaming the variable, we do not expect significant code change.

amcastro-tri, 2019-07-23:
Since we'll rename, can I propose we use `grad_h_nm_M`? with lowercase for `nm` and just the "expressed-in" frame in upper case.

DamrongGuoy, 2020-02-25:
No need anymore. `ContactSurface` does not provide ∇h anymore.

closed 2020-02-25 by DamrongGuoy