RobotLocomotion/drake #11831: Add a bushing force element (as a low-budget loop weld joint)

sherm1, 2019-07-25:
Per issue #9566 we need to support topological loops in the multibody system graph. This requires a spanning tree plus constraints to close the loops. It is sufficient just to cut bodies and weld them back together with a weld constraint, but currently we don't have hard constraints in MBP.

For spiral 0, a bushing force element (used to keep a frame on body A nearly coincident with a frame on body B) can be used as a cheap, penalty-method approximation of a weld constraint. This can be added using MBP's ForceElement, which already has LinearSpringDamper as a 1dof example. A bushing is a generalization of that.

Implementation notes per discussions with @mitiguy and @amcastro-tri:
- from MBP kinematics we get X_AB and V_AB
- decompose R_AB into roll-pitch-yaw RPY angles
- use RPY and w_AB to calculate d/dt RPY (probably add method to RollPitchYaw class)
- put spring-damper forces k*R+c*Rdot etc. along each of the 6 axes
- accumulate to apply wrench to bodies A and B
- must also compute potential energy, conservative power, non-conservative power
- will need to figure out how to solve efficiently -- likely need implicit integrator or modification to implicit Stribeck to include penalty weld joints

Note that this is a fully nonlinear force element, but it is only a reasonable model for a rubber bushing when the RPY angles are near zero (otherwise the stiffness can't be decomposed neatly into independent axes).

Having this element and the ability to manually add it to an MBP system would at least allow us to model systems with topological loops, with various caveats. Subsequent spirals can take care of adding the weld joint automatically, first the penalty one and then the hard constraint; that's not part of this particular issue.

Paul expressed interest in adding this force element so I'm assigning it initially to @mitiguy.

Thoughts?

sherm1, 2019-07-25:
BTW Simbody has exactly the bushing element we would need:
- [API Doxygen](https://simbody.github.io/simbody-3.6-doxygen/api/classSimTK_1_1Force_1_1LinearBushing.html#details)
- [guts of the code](https://github.com/simbody/simbody/blob/master/Simbody/src/Force_LinearBushing.cpp#L403)

sherm1, 2019-07-25:
Per discussion with @amcastro-tri, suggested approach:
- Draft working bushing force element, possibly missing functionality and with inadequate docs and unit tests
- add a simple example that shows how to use it (e.g. a four-bar linkage)
- Push that branch so we can experiment with it to see if it at least partially addresses users who must have loops
- Then return to the branch and polish the PR to get the code into master.

amcastro-tri, 2019-07-26:
cc'ing @RussTedrake, @mposa

mposa, 2019-07-26:
Thanks for taking a look at this. Here are some comments:

- For our purposes, we do not need the full 6DOF weld. A 3DOF ball joint (xyz and not rpy) is sufficient.
-Right now, we're actually using the RBT `addDistanceConstraint` (https://github.com/RobotLocomotion/drake/blob/master/attic/multibody/rigid_body_tree.cc#L3224) to represent massless loop joints (1DOF).
- All of these can just be thought of as kinematic constraints `y(q)=0`, where the restoring spring-damper force is
-Lower DOF constraints means the unconstrained model also needs fewer DOFs, which strikes me as a good thing.
```
J = dy/dq;
F = -J^T (Kp*y + Kd*ydot
```
Is there any reason not to implement this as an virtual class, with the weld as Example 1, where users could implement their own versions (ball joints, distance constraints, etc.)?

sherm1, 2019-07-26:
The `multibody::ForceElement` we're planning to use here is just the virtual class you have in mind, I think. There is already a point-to-point spring/damper example there (enforcing a -1-dof distance constraint) and the bushing would be the other extreme (enforcing a -6-dof weld constraint). However, force elements don't introduce any new equations (they just add to the already-present rhs) so may not need to be trimmed down to the minimum size when used as "penalty" constraints.

Of course this is no substitute for real constraints in MBP, which are on the near-term roadmap, but this might be a quick way to make progress in the meanwhile.

mposa, 2019-07-26:
The only thing that's stopped me from just using `ForceElement` and handling the constraints myself is your comment above

> will need to figure out how to solve efficiently -- likely need implicit integrator or modification to implicit Stribeck to include penalty weld joints

Will the discrete time integrator play nicely with just adding stiff spring forces? If so, then, from our end, we can probably just implement a custom loop closure until you are ready for an official constraint definition.

sherm1, 2019-07-26:
Yep, we don't know the answer yet. We'll try that as soon as we can. It will cause trouble with the spring stiffnesses set very high I'm sure. It may require an addition to the discrete solver to include the bushing in the implicit portion (@amcastro-tri already did that for joint stops so I think it is fairly straightforward?).

EricCousineau-TRI, 2019-10-28:
FTR, there is a Stack Overflow question about this:
https://stackoverflow.com/questions/58559987/constraint-force-for-closed-loop-multibodyplant

RussTedrake, 2020-04-08:
@sherm1 , @mitiguy -- safe to close this one now, right?

sherm1, 2020-04-08:
Closed via PR #11946, with energy and power methods TBD covered in issue #12982.

closed 2020-04-08 by sherm1

jwnimmer-tri, 2020-04-08:
I would like to see parser support for this (`<drake:linear_bushing_roll_pitch_yaw>`) before calling the story complete.  I don't see an issue for that yet?  Even in Anzu we're substantially less likely to use it without parser support.

RussTedrake, 2020-04-08:
I was thinking that it would not be a new tag, but that it would be added automatically if the sdf contained a closed kinematic chain.  Had a conversation with @joemasterjohn along those lines.

jwnimmer-tri, 2020-04-08:
Fine by me, whatever mechanism makes sense.

sherm1, 2020-04-08:
Having an explicit tag makes sense even if we do add them automatically also. I'll open an issue for that.

sherm1, 2020-04-08:
-> #13021 (for the parsing tag, not automatic bushing generation)