RobotLocomotion/drake #11850: Issues in using Signal-Scope probably due to different instances of LCM installation

zulfiz, 2019-07-26:
Hi,

I am using Drake for building my project. I am using an older version of Drake that uses bazel version 0.19. Reasons are as all the code written for our project is compatible with that version of drake. We know that, Drake sets up its own version of LCM. I think for me it got installed under /opt/lcm/1.3.95.20180523. Now, in order to visualize the  lcm messages from our robot, I downloaded and installed lcm-signal-scope. I prompted me to download lcm as it wasn't able to find it. Therefore, I installed lcm-1.3.1 on my system. But, now when I run the signal scope it isn't able to decode incoming lcm signals. It throws the following error:

```
File "<string>", line 143, in decodeMessageFunction
KeyError: 'R\xff\xd3\tO\x1a\x9b\x1f'

```

My system is Ubuntu 16.04. Any help in resolving this issue will be highly appreciated. Thank you!

sherm1, 2019-07-28:
Can anyone advise @zulfiz on this?

sherm1, 2019-07-28:
Assigning @jamiesnape as a first guess at someone who might be able to help -- please feel free to reassign.

EricCousineau-TRI, 2019-08-01:
Are you able to upgrade to a later version of Drake first? That'd make things simpler to address for us.

EricCousineau-TRI, 2019-12-12:
Closing this for now due to inactivity. Please let us know if you can upgrade, or if there are other constraints.

closed 2019-12-12 by EricCousineau-TRI