RobotLocomotion/drake #11857: Possibility of adding a joint group tag in the sdf/urdf of Drake

huihuaTRI, 2019-07-29:
During the Drake developer meeting, we have discussed the possibility to add tags which could put certain joints into a group in the urdf or sdf. The purpose is to easily grab/set state or command for certain joints. For example, the joints for the left arm or just the torso. It turns out that such tags are not available in urdf/sdf. 

The question is that is it possible to add those customized tags in urdf/sdf and get processed by Drake. 

One way to work around this is to just create several leaf systems to do this kind of tasks specifically.

sherm1, 2019-07-29:
If this isn't already in urdf/sdf this would be a novel Drake feature. Can you explain what it would be used for and why a user couldn't do it already using joint names (for example)?

jwnimmer-tri, 2019-07-29:
I would argue that in any case, the correct process would be to design and implement and validate this feature in TRI-internal repository (aka Anzu) first to de-risk its requirements, and only if and when we've built up that experience with the solution space to contribute it back to Drake.

huihuaTRI, 2019-07-30:
To answer Sherm's question, an example case would be wanting to control a particular arm of a mobile base robot with more than one arm, while keeping the rest of the body static. This is a feature in Moveit!, in which you can generate a trajectory plan for arbitrary limbs.

It is good to have, but not a necessary feature.

EricCousineau-TRI, 2019-08-12:
Can you point to MoveIt! documentation that embeds these tags within URDF? I thought they ultimately used a separate config file for handling joint groupings and all that?

huihuaTRI, 2019-09-19:
I have limited experience with MoveIt! also. I tend to close this issue for now since I already have some workaround solution for my problem.

closed 2019-09-19 by huihuaTRI

EricCousineau-TRI, 2019-09-19:
Just out of curiosity, can you mention what your workaround was?

huihuaTRI, 2019-09-19:
It's a very tedious way. I just create controllers for different parts and list the joint names of that part explicitly.