RobotLocomotion/drake #11868: All python tests fail under python3-dbg

jamiesnape, 2019-07-30:
Assertion fails within `python3-dbg` version 3.6.8 on Bionic:

```
python3-dbg: ../Modules/gcmodule.c:380: visit_decref: Assertion `_PyGCHead_REFS(gc) != 0' failed.
external/bazel_tools/tools/test/test-setup.sh: line 310: 11800 Aborted                 (core dumped) "${TEST_PATH}" "$@" 2>&1
```

jamiesnape, 2019-07-30:
Actually, not all tests, but the following:
```
//bindings/pydrake:py/all_test
//bindings/pydrake:py/automotive_test
//bindings/pydrake:py/geometry_test
//bindings/pydrake:py/perception_test
//bindings/pydrake/attic:py/attic_forwarding_test
//bindings/pydrake/attic/multibody:py/rigid_body_plant_test
//bindings/pydrake/attic/systems:py/controllers_test
//bindings/pydrake/attic/systems:py/sensors_test
//bindings/pydrake/doc:gen_sphinx_test
//bindings/pydrake/examples:py/acrobot_test
//bindings/pydrake/examples:py/compass_gait_test
//bindings/pydrake/examples:py/manipulation_station_test
//bindings/pydrake/examples:py/pendulum_test
//bindings/pydrake/examples:py/quadrotor_test
//bindings/pydrake/examples:py/rimless_wheel_test
//bindings/pydrake/examples:py/van_der_pol_test
//bindings/pydrake/examples/multibody:cart_pole_passive_simulation_test
//bindings/pydrake/examples/multibody:pendulum_lqr_monte_carlo_analysis_test
//bindings/pydrake/maliput:py/maliput_test
//bindings/pydrake/manipulation:py/planner_test
//bindings/pydrake/manipulation:py/simple_ui_test
//bindings/pydrake/multibody:py/inverse_kinematics_test
//bindings/pydrake/multibody:py/parsing_test
//bindings/pydrake/multibody:py/plant_test
//bindings/pydrake/solvers:py/ipopt_solver_test
//bindings/pydrake/solvers:py/mathematicalprogram_test
//bindings/pydrake/solvers:py/nlopt_solver_test
//bindings/pydrake/solvers:py/osqp_solver_test
//bindings/pydrake/solvers:py/scs_solver_test
//bindings/pydrake/systems:drawing_graphviz_example_test
//bindings/pydrake/systems:py/controllers_test
//bindings/pydrake/systems:py/custom_test
//bindings/pydrake/systems:py/general_test
//bindings/pydrake/systems:py/lcm_test
//bindings/pydrake/systems:py/lifetime_test
//bindings/pydrake/systems:py/meshcat_visualizer_test
//bindings/pydrake/systems:py/monte_carlo_test
//bindings/pydrake/systems:py/perception_test
//bindings/pydrake/systems:py/primitives_test
//bindings/pydrake/systems:py/rendering_test
//bindings/pydrake/systems:py/scalar_conversion_test
//bindings/pydrake/systems:py/sensors_test
//bindings/pydrake/systems:py/trajectory_optimization_test
//bindings/pydrake/systems:py/value_test
//examples/manipulation_station:end_effector_teleop_mouse_test
//examples/manipulation_station:end_effector_teleop_sliders_test
//examples/manipulation_station:joint_teleop_test
//examples/manipulation_station:print_station_context_test
//examples/van_der_pol:plot_limit_cycle_test
//manipulation/util:py/geometry_inspector_test
//manipulation/util:py/show_model_test
```

EricCousineau-TRI, 2019-07-30:
Can reproduce this error if you just have a single file that just imports an offending module:
```py
import pydrake.systems.framework
```

(no code necessary)

EricCousineau-TRI, 2019-07-30:
Yup, confirmed that it's the `PYDRAKE_PREVENT_PYTHON3_MODULE_REIMPORT` hack:
b9d060b5d

@jamiesnape Can you try to see if there's simple enough workaround / fix for this? If not, I can look into more on Thursday.

EricCousineau-TRI, 2019-07-30:
If we can't find a workaround, we can always re-engineer the imports to not be cyclic. First part would be address the TODO to move `Value<>` stuff outside of `pydrake.systems.framework` and into `pydrake.common`.

jamiesnape, 2019-07-31:
> @jamiesnape Can you try to see if there's simple enough workaround / fix for this? If not, I can look into more on Thursday.

Sure, let me catch up with the history of that code and I will get back to you.

jamiesnape, 2019-07-31:
> If we can't find a workaround, we can always re-engineer the imports to not be cyclic.

I think it makes sense to do this anyway. I don't think I have a workaround that I am confident does not have unintended side effects.

EricCousineau-TRI, 2019-08-08:
FTR, this workaround suppresses the warning:
42ed5c513

PR #11910 (2019-08-08): pydrake: Work around refcounting issue with cyclic C extension import
Resolves #11868

This should resolve the issue. While it's not great, it's a 5min fix.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11910)
commit 07a14b0f (2019-08-08): pydrake: Work around refcounting issue with cyclic C extension import (#11910)

Resolves #11868
closed 2019-08-08 by jamiesnape