RobotLocomotion/drake #11880: camera_properties: Support setting with intrinsic matrix, possibly using CameraInfo

EricCousineau-TRI, 2019-07-31:
Would like this to supersede #8850. @SeanCurtis-TRI I've assigned you for now given that it supersedes that issue, but can reassign if you'd like. (I could also reword that issue if you'd like.) 

Regarding this TODO comment:
https://github.com/RobotLocomotion/drake/blob/594f1864b5ef654e1b4c4eaf0b2ef9a58e8334eb/geometry/render/camera_properties.h#L13-L16

Potential solutions could be actually using `CameraInfo`, or making a static factory method or something that takes an intrinsic matrix.

For renderer support, I'm fine if a renderer currently chokes if it can't support a set of intrinsics (either a non-central principal point, unequal fx + fy, skew, etc.).

\cc @thduynguyen Part of this could also involve adding back-projection utilities like what you requested, and then possibly have Anzu use this structure?

SeanCurtis-TRI, 2020-06-12:
Ok, I'm ready to start submitting PRs against this. The effort to do this is going to come in multiple phases. The overall result can be found here: https://github.com/SeanCurtis-TRI/drake/tree/PR_camera_intrinsics

Here's the plan:

- [ ] Submit changes to `CameraInfo`, `RgbdSensor` and introduce `RenderCameraProperties`
    - This provides the core data structures for specifying the intrinsics and API for passing them to the `RgbdSensor`. However, at this point, actual intrinsic values that aren't "centered and symmetric" will spew a warning.
- [ ] Implement the `geometry/` infrastructure to support setting camera intrinsics. This will hit `QueryObject`, `GeometryState` and, the `RenderEngine` base class.
    - The `RenderEngine` base class will include behavior such that if a derived class hasn't implemented the behavior necessary to set the camera from intrinsics, it will default to the simple version we have today.
- [ ] Implement the use of full camera intrinsics in `RenderEngineVtk`. This will make the feature live and accessible.

------ everything above the line implemented in the linked branch; everything below still to be done -----

- [ ] Deprecate all of the old `CameraProperties` and `DepthCameraProperties` related APIs.
- [ ] Add intrinsics to the Ospray render engine.
- [ ] Add intrinsics to the `RenderEngineGl` that is currently porting over.
- [ ] After the deprecation period, kill the old API entirely.

EricCousineau-TRI, 2020-11-19:
Woot for the PR :bullettrain_front:!

PR #14376 (2020-11-21): [render] Deprecate CameraProperties and DepthCameraProperties
This deprecates the old camera properties classes themselves. As part of that endeavor:

  - bindings deprecate the constructors.
  - A slew of deprecation warning suppressions everywhere.
    - In some cases, test cases were moved around to facilitate deprecation
      suppression en masse (see render_camera_test.cc)
  - Updated python tests to expect deprecation warnings.

Towards resolving checklist in #11880.

Closes #11880

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/14376)
closed 2020-11-21 by sherm1