RobotLocomotion/drake #11883: Upstream scs release vendors stale, incompatible version of qdldl

jwnimmer-tri, 2019-08-01:
Drake master is currently using **[SCS](https://github.com/cvxgrp/scs) v2.0.2** from Nov 30, 2017.  We would like to upgrade to the next (and most recent) release **SCS 2.1.1** from Jul 8, 2019, per our policy of always using the latest upstream releases.  (See https://github.com/cvxgrp/scs/releases page for the list of SCS releases.)

During the past 21 months since the last release, SCS has updated from using `linsys/direct/external/ldl.h` in v2.0.2 to `linsys/direct/external/qdldl/qdldl.h` in 2.1.1.

Separately, Drake also depends on [OSQP](https://github.com/oxfordcontrol/osqp) which itself uses [QDLDL](https://github.com/oxfordcontrol/qdldl/).

The problem is that SCS has vendored a stale, incompatible copy of QDLDL into their externals folder.  Building SCS against the latest QDLDL used by OSQP does not work.  I have not tried building two different copies of QDLDL into Drake at once, but I suspect it would not work.

We should find some way to upgrade to the latest SCS.  I suspect we will have to submit a PR to SCS to vendor and use the latest QDLDL, have them release a new version number of SCS, and then upgrade Drake to that new version, skipping 2.1.1 entirely.

My WIP branch for the SCS upgrade is at https://github.com/jwnimmer-tri/drake/commits/new-release-scs, as a starting point for this work.

\CC @EricCousineau-TRI FYI

jwnimmer-tri, 2019-09-06:
@jamiesnape @EricCousineau-TRI Can I nudge this a bit higher on the priority list?  Maybe target having some progress prior to the Oct 1st upgrade cycle?

Note that OSQP has releases 0.6.0 now (Drake is at 0.5.0).  I'm planning on leaving the 0.6.0 upgrade to be resolved concurrently with this issue, in case they end up being intertwined.

EricCousineau-TRI, 2019-09-09:
We've got a few other things to deprecate / upgrade for Oct 1, as well as some other items (e.g. OSPRay).
I'm hesitant to up this in priority if it's just for cleanliness. Would this be blocking any significant features? (e.g. using a good optimizer that we need upgrades for?)

jwnimmer-tri, 2019-09-09:
It's just a question of how long we want to live on a stale pin of two solvers.  OSQP is relatively prominent in our choose_best_solver matrix.

I will put this back on my radar to work in a week or two, in case other priorities have taken over Jamie's time.  I agree that the work to cull the OS / Py2 support matrix is more important.

EricCousineau-TRI, 2019-09-09:
SGTM, thanks!

PR #12084 (2019-09-25): workspace: Update scs to latest release
Closes #11883.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/12084)
closed 2019-09-25 by jwnimmer-tri