RobotLocomotion/drake #11885: pydrake module deprecation dates 2019-06-01

jwnimmer-tri, 2019-08-01:
We have some lying deprecation dates on master:
```
./bindings/pydrake/systems/_controllers_extra.py:6:    _forward_callables_as_deprecated(locals(), _attic, "2019-06-01")
./bindings/pydrake/systems/_sensors_extra.py:6:    _forward_callables_as_deprecated(locals(), _attic, "2019-06-01")
```

I thought there was an issue about repairing some module deprecation warnings to actually trigger, at which point we could reset these deprecation dates to be 3 months past that fix.

On a quick search though, I was unable to find that issue.

EricCousineau-TRI, 2019-08-05:
Yeah, there was #11363, where I thought that I had fully resolved attic-forwarding. This seems like it falls under that issue, but wasn't caught in my debugging.

Will look into this.

EricCousineau-TRI, 2019-08-05:
Er, wait - was it that I didn't fix those dates, or that these deprecations don't trigger?

jwnimmer-tri, 2019-08-05:
My recollection was on that 2019-06-01, the warnings were not triggering, so we kept the methods (possibly after a PR merge + revert once we learned).  I don't know if the warnings trigger now.  If they do, we can update the dates to 2019-11-01 to start a new timer.

EricCousineau-TRI, 2019-08-05:
I'll check, but the way these are implemented (using `_deprecated_callable`), these should always have thrown deprecation warnings.

I'm checking the unittest to see if it fails if I mess up `_deprecated_callable`. I will also checkout pre-merge of #11645 (or just revert it) to ensure that it still fails.

If that's the case, then I'd say that these deprecations never were falsely missed, and should be scrubbed as is?

jwnimmer-tri, 2019-08-05:
Yes, if these have been yelling for at least three months, then I think it's OK to purge the code now.

EricCousineau-TRI, 2019-08-05:
Confirmed the unittest fails if there's no deprecation: 454ceffc5
Also confirmed that this same change fails with #11645 also reverted: 7403292de

So yeah, I'll go ahead and scrub 'em.

PR #11905 (2019-08-06): py systems: Remove 2019-06-01 deprecations
Resolves #11885

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11905)
commit 0dfec73f (2019-08-06): py systems: Remove 2019-06-01 deprecations

Resolves #11885
closed 2019-08-06 by EricCousineau-TRI