RobotLocomotion/drake #11888: Remove Isometry3 from systems/rendering

SeanCurtis-TRI, 2019-08-01:
The `systems/rendering` folder still has a number of artifacts that all depend on `Isometry3` -- this needs to be scrubbed as well.

SeanCurtis-TRI, 2020-05-13:
NOTE: This impacts `geometry/` because `SceneGraph` uses the `PoseBundle` class to communicate with `drake_visualizer`. If we created a single `drake_visualizer` system that connected to `SceneGraph`'s `QueryObject` output port then this output port could simply be eliminated.

jwnimmer-tri, 2020-05-13:
FWIW it would be super quick to just change PoseBundle to use RigidTransform -- deprecate a getter/setter pair and add new ones.  Probably a 200 LOC PR.

SeanCurtis-TRI, 2020-05-13:
Thinking deprecate `set/get_pose` and adding `set/get_transform`?

jwnimmer-tri, 2020-05-13:
Exactly.  (I didn't actually think about the names, but I think that's a good name.)

PR #13327 (2020-05-18): Deprecate the Isometry3 in systems/rendering
 - PoseBundle::{set|get)_pose deprecated; replaced with (set_)transform()
   - get_pose() no longer returns a reference to an Isometry3.
 - PoseVector::get_isometry deprecated; replaced with transform()
   - Also added set_transform() based on old TODO.
 - Multiple downstream consumers (and their tests) updated to use new API.
 - Bindings updated (with deprecations) and test
   - pose_bundle_to_draw_message_test

Resolves #11888

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13327)
closed 2020-05-18 by sammy-tri