RobotLocomotion/drake #11894: IPOPT Error when Using Drake from Python

bsbretly, 2019-08-02:
Hello,

I am following the instructions [here](https://drake.mit.edu/python_bindings.html#using-python-bindings) to install the python binding for drake. While building the python bindings, I encounter the following error:

`INFO: Call stack for the definition of repository 'ipopt' which is a pkg_config_repository (rule definition at /home/brett/drake-build/_bazel_brett/4db2c1afdf221ed1a5ae63ad68dbfb3d/external/drake/tools/workspace/pkg_config.bzl:262:25):
 - /home/brett/drake-build/_bazel_brett/4db2c1afdf221ed1a5ae63ad68dbfb3d/external/drake/tools/workspace/ipopt/repository.bzl:20:5
 - /home/brett/drake/tools/workspace/default.bzl:158:9
 - /home/brett/drake/WORKSPACE:10:1
ERROR: An error occurred during the fetch of repository 'ipopt':
   Unable to complete pkg-config setup for @ipopt repository: error 1 from [/usr/bin/pkg-config, "ipopt"]:'

Thank you for your help!
Brett

jwnimmer-tri, 2019-08-02:
Please post exactly the list of command(s) you are running that produce that error.

If you are compiling Drake, my first guess is that `setup/ubuntu/source_distribution/install_prereqs.sh` was forgotten, or did not complete successfully.

From the issue template:
> * Have you rerun `install_prereqs.sh` on your current commit of `drake`?

bsbretly, 2019-08-02:
@jwnimmer-tri - thanks for the quick response. I have updated my original post with the information you requested.

jwnimmer-tri, 2019-08-02:
Great, thanks!

Here's one problem: the `/opt/drake/share/drake/setup/install_prereqs` script is intended for users who have downloaded a pre-compiled Drake release and who want to just import and run the precompiled release.  It only installs the runtime depenencies, not the build-time dependencies.

To install build-time dependencies (i.e., to be able to to run CMake build of Drake), you'd need to run `setup/ubuntu/source_distribution/install_prereqs.sh` from the source tree.

bsbretly, 2019-08-04:
@jwnimmer-tri - thank you! Running the script at `setup/ubuntu/source_distribution/install_prereqs.sh` was my problem - I no longer get the IPOPT issue.

I am, however, still unable to complete the build. Maybe it's easier to just include a screen grab of my terminal showing the error:
[image]

As you can see, I tried to build again - still with no success. If it's more appropriate to make a new issue for this problem, I can happily do that as well. 

Thanks again!

jwnimmer-tri, 2019-08-04:
Per https://drake.mit.edu/getting_help.html#asking-your-question we'd usually do this help on StackOverflow, but we can work it here also for now.

_"Internal compiler error: Killed"_ is usually an out-of-memory condition due to too-high parallelism.

To reduce parallelism, create a file `$HOME/.bazelrc` that contains a line like `build --jobs=2` or similar, to limit the number of CPUs used in parallel (so as to reduce memory pressure).  I would start with a value around 80% of your total vCPUs and work your way down until it succeeds.

bsbretly, 2019-08-04:
Many thanks @jwnimmer-tri - your super quick feedback and patience is much appreciated. Everything has built - I will close this issue.

closed 2019-08-04 by bsbretly