RobotLocomotion/drake #11896: LcmSubscriberSystem: magic_number_ not a condition variable

jamiesnape, 2019-08-02:
Relates to this code:

https://github.com/RobotLocomotion/drake/blob/0f43c61a99d7f20cbf8d99308f4b034ffecb67ca/systems/lcm/lcm_subscriber_system.cc#L71-L74

Example report from running DRD:
```
[ RUN      ] ValkyrieSimulationTest.TestIfRuns
==17== not a condition variable: cond 0xac71830
==17==    at 0x4C3ACA4: pthread_cond_destroy_intercept (drd_pthread_intercepts.c:1063)
==17==    by 0x4C3ACA4: pthread_cond_destroy@* (drd_pthread_intercepts.c:1073)
==17==    by 0x940B64: drake::systems::lcm::LcmSubscriberSystem::~LcmSubscriberSystem() (lcm_subscriber_system.cc:74)
==17==    by 0x940B9D: drake::systems::lcm::LcmSubscriberSystem::~LcmSubscriberSystem() (lcm_subscriber_system.cc:71)
==17==    by 0x1DF0CA: std::unique_ptr<drake::systems::System<double>, std::default_delete<drake::systems::System<double> > >::~unique_ptr() (unique_ptr.h:268)
==17==    by 0x1E3194: std::vector<std::unique_ptr<drake::systems::System<double>, std::default_delete<drake::systems::System<double> > >, std::allocator<std::unique_ptr<drake::systems::System<double>, std::default_delete<drake::systems::System<double> > > > >::pop_back() (stl_vector.h:979)
==17==    by 0x1DF5C7: drake::systems::internal::OwnedSystems<double>::~OwnedSystems() (diagram.h:46)
==17==    by 0x9774FB: drake::systems::Diagram<double>::~Diagram() (diagram.h:91)
==17==    by 0x19DB20: drake::examples::valkyrie::ValkyrieSimulationTest_TestIfRuns_Test::TestBody() (valkyrie_simulation_test.cc:52)
==17==    by 0xAB4659: void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) (gtest.cc:2479)
==17==    by 0xAA7925: testing::Test::Run() (gtest.cc:2517)
==17==    by 0xAA8013: testing::TestInfo::Run() (gtest.cc:2693)
==17==    by 0xAA8481: testing::TestCase::Run() (gtest.cc:2811)
==17==    by 0xAAFD32: testing::internal::UnitTestImpl::RunAllTests() (gtest.cc:5177)
==17==    by 0xAB5AD9: bool testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) (gtest.cc:2479)
==17==    by 0xAAFAFF: testing::UnitTest::Run() (gtest.cc:4786)
==17==    by 0xA87448: main (drake_cc_googletest_main.cc:32)
```
While the DRD jobs are not on CI, the defect report seems reasonable enough.

jwnimmer-tri, 2020-02-03:
As of #12664, Valkyrie code no longer exists on master. Might be time to just close out this issue.

jwnimmer-tri, 2020-04-21:
Seems likely to be unreproducible now, without Valkyrie.

closed 2020-04-21 by jwnimmer-tri