RobotLocomotion/drake #11906: Generate box's volume mesh with regid core at medial axis or offset surface.

DamrongGuoy, 2019-08-06:
Extend MakeBoxVolumeMesh() to include its medial axis or its offset surface.

Given a box [-hx, hx]x[-hy,hy]x[-hz,hz], where h = (hx,hy,hz) is its corner vector. We can describe its offset surface at distance d as another box [-sx, sx]x[-sy, sy]x[-sz, sz], where
             s = (sx, sy, sz) = h - (d,d,d)
with the assumption that d <= min {hx, hy, hz}.

A Cartesian grid with vertices {-hx, -sx, sx, hx} x {-hy, -sy, sy, hy} x {-hz, -sz, sz, hz} can have the offset surface in the mesh.

In the case that d = min {hx, hy, hz}, one or two or three of sx, sy, sz become zero, and the offset surface becomes the medial surface, the medial axis, or the origin.

Instead of uniform sampling using LinSpaced on [-hx,hx], we'll do it on each of [-hx,-sx],[-sx,sx], and [sx,hx]. Similarly for y and z.

sherm1, 2019-08-06:
Sounds great, thanks @DamrongGuoy. A few unresolved questions (maybe need separate issues?):
- whether we ought to mesh differently if we know the ε=1 boundary is a face rather than the medial surface
- whether the surface triangle mesh should be generated along with the tet mesh (seems like it would be very cheap as a side effect and would let us use a box as a rigid object)
- how to convey the location of the ε=1 boundary to downstream code -- if we generate boundary-condition-aware tet meshes, should the boundary vertices already carry their pressure field values?

sherm1, 2019-08-06:
cc @ryanelandt

amcastro-tri, 2019-08-06:
I agree with this issue, good observation. However I am surprised to see the "medium" priority of this issue while #11154 doesn't even have a priority level, which I believe in order to satisfy our sim needs should be higher?

DamrongGuoy, 2019-08-06:
@amcastro-tri  Thanks for asking about setting priority label. I forgot about it. I did it for #11154 now.

ryanelandt, 2019-08-07:
> * whether the surface triangle mesh should be generated along with the tet mesh (seems like it would be very cheap as a side effect and would let us use a box as a rigid object)

Alternatively, you could write a general function that creates rigid meshes from compliant meshes. This function works as follows: initialize an empty tri mesh, loop over each tet, if ε=0 on three of the vertices add this face to the tri mesh, return tri mesh.

DamrongGuoy, 2020-10-15:
I have PR #14216 is in review now. It will allow us to have these meshes and pressure fields.

[image]

DamrongGuoy, 2020-10-22:
PR #14216 is merged now. Next I'll enable usage of this mesh in hydroelastics_internal and we'll finish.

DamrongGuoy, 2020-11-04:
PR #14251 has been merged, so we can close this issue now.

closed 2020-11-04 by DamrongGuoy

EricCousineau-TRI, 2020-11-04:
Sweet - thanks!!!