RobotLocomotion/drake #11908: Radau1 should behave exactly the same as Implicit Euler.

amcastro-tri, 2019-08-07:
With @antequ-TRI we run both Radau1 and implicit Euler (IE) integrators on a same problem, in fixed step mode. While Radau1 in theory is exactly the same as the implicit Euler, we observed both of them to behave differently. We that we mean that, the number of Newton-Raphson iterations, residuals and the solutions themselves were not the same. We expect solutions and residuals to be within machine epsilon since both integrators should essentially follow the same path.

For more complex problems with error control we observed the integrators not to behave as theoretically expected. However we could not analyze further given that Radau1 and IE are so different that we still are unsure if there still is a bug to be uncovered.

antequ-TRI, 2019-08-07:
@edrumwri , I made a commit here that produces the repro:
https://github.com/antequ-TRI/drake/commit/2f050e030408af2813bb849dad53398c4c78c45a

This commit changes a few things:
1) a minor change to IntegratorBase to allow us to set target accuracy during fixed step mode. This allows us to make the radau1 newton-raphson tolerance match implicit euler newton-raphson tolerance in fixed mode
2) a minor change to implicit_euler_integrator-inl.h to disable the implicit trapezoid during fixed mode (so we know failures don't come from the trapezoid)
3) a new leafsystem, called BoxPlant, that calculates the stribeck friction force between the box and a fictional plane. See examples/box/box_plant.h . The state change is in DoCalcVectorTimeDerivatives and the friction expression is in CalcFrictionFromVelocity.
4) an example, called PassiveSimulation (passive_simulation.cc), that can run this with various integrators and visualization settings, and output the results to a file.

To run the system (with friction), use:
```
bazel run //examples/box:passive_simulation  -- --integration_scheme='implicit_euler' --fixed_step=true --run_filename='boxout_ie' --meta_filename='boxmeta_ie'
bazel run //examples/box:passive_simulation  -- --integration_scheme='radau1' --fixed_step=true --run_filename='boxout_radau1' --meta_filename='boxmeta_radau1'
```
All outputs are saved to bazel-bin/examples/box/passive_simulation.runfiles/drake
The states (and frictional forces) are located in boxout_ie.csv for implicit euler and boxout_radau1.csv for Radau 1. 

The metainformation (step sizes, number of derivative evaluations) are located in boxmeta_ie.csv for implicit euler and boxmeta_radau1.csv for Radau 1. In my run, it took 1447 steps for implicit euler (7536 derivative evals) while it took 1451 steps for radau 1 (8602 derivative evals).

To run it without friction, use:
```
bazel run //examples/box:passive_simulation  -- --integration_scheme='implicit_euler' --fixed_step=true --box_mu_s=0. --run_filename='boxout_ie_nf' --meta_filename='boxmeta_ie_nf'
bazel run //examples/box:passive_simulation  -- --integration_scheme='radau1' --fixed_step=true --box_mu_s=0. --run_filename='boxout_radau1_nf' --meta_filename='boxmeta_radau1_nf'
```
These runs will also save the output, with a "_nf" postpended to the filename before ".csf". In my runs, both integrators took 1440 steps and 4323 derivative evals. 

The other settings you can play with are:
 `max_time_step=3e-3` for timestep size,
`fixed_tolerance=1e-5` for newton-raphson tolerance,
`target_realtime_rate=1.0` for simulation rate,
`simulation_time=4.0` for simulation duration, and
`visualize=true` for whether to turn on the visualizer

One other difference between the integrators is radau1 passes in x0 into maybefreshenmatrices in this line:
```
  // Calculate Jacobian and iteration matrices (and factorizations), as needed.
  if (!this->MaybeFreshenMatrices(t0, xt0, h, trial, construct_iteration_matrix,
      &iteration_matrix_radau3_)) {
```
while implicit euler passes xtplus (which is the most recent iterate from the previous trial) into maybefreshenmatrices in this line:
```
  if (!this->MaybeFreshenMatrices(tf, *xtplus, h, trial,
      compute_and_factor_iteration_matrix, &iteration_matrix_)) {
```

I tried modifying them to make that line match, and somehow it still does not fix the difference in behavior.

edrumwri, 2019-08-07:
This is great, Ante! Thank you for the commit and also the detailed description of your modifications.

PR #12088 (2019-10-02): Euler radau reconcile
Reconciles the disparity between implicit Euler and 1-stage Radau, using Ante's box example.
Resolves #11908 and #11939.

Notes for reviewers:
1. The box example is not part of this PR (it was just the example that Ante discovered this problem with- essentially any example should exhibit this behavior).
2. The PR moves all of the unit tests from implicit Euler to "implicit_integrator_test.h", allowing the Radau and implicit Euler integrators to share the unit tests.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/12088)
closed 2019-10-02 by edrumwri