RobotLocomotion/drake #11913: models: Should migrate data to `RobotLocomotion/models` per README advice

EricCousineau-TRI, 2019-08-08:
Follow-up to #10811.

EricCousineau-TRI, 2020-04-10:
FYI @IanTheEngineer

jwnimmer-tri, 2020-05-04:
This is confusing a few people every month, so I'm going to work on it.

jwnimmer-tri, 2020-06-24:
\CC @IanTheEngineer

jwnimmer-tri, 2021-08-02:
It's possible that the resolution of #13942 will end up being a prerequisite for this work.  If we decide that the sdformat / urdf files should move as well, it probably makes sense to do them as part of the same file relocation as the meshes.

jwnimmer-tri, 2023-04-11:
The only remaining one to port is the PR2. It's complicated by the fact that we're currently reprocessing those models with `stl2obj` and then installing the reprocessed copies. The easy fix is to commit the reprocessed files into the models repository. It might be worth seeing if there is any a more elegant solution available instead.

jwnimmer-tri, 2024-03-19:
I'll roll the one remaining task into #13942, so we can close this one.

closed 2024-03-19 by jwnimmer-tri