RobotLocomotion/drake #11917: Allow Parameters in MultiBodyPlant to be symbolic for use in System Identification

tarquasso, 2019-08-09:
Parameters in MultiBodyPlant (MBP) can only be double, not symbolic. The use of MBP for system identification of a robot manipulator would be very useful. System Identification already exists in Drake, it provides TrigPoly enables proper lumped parameter structuring. What is left is that MBP's parameters such as inertia H and coriolis/centrifugal C cannot be symbolically constructed from a MBP.

More details by @jwnimmer-tri  :
`m` and `l` could probably be a Variable  - though the way the example's main program is written would have to be tweaked a litte. The `g` wouldn't compile as Variable right now but at first glance looks like a simple fix. We'd also have to strip off the visual geometry for this to work (can't have symbolic visual geometry frame offsets). The `damping` is hard-coded to be a double. Making the various parameters type T rather than double should be all that's required. To transmogrify easily, the parameters have to move into the context, instead of being direct member fields of the, e.g., joints or whatever.  And so their getters have to take a new Context argument now, etc. You can make symbolic plants without doing all of that work, but working with them becomes much more difficult - ok for system ID purposes...

RussTedrake, 2021-04-01:
@joemasterjohn -- this is mostly done (with a few notable exceptions), and seems to not be used for tracking.  Perhaps you can close it?

joemasterjohn, 2021-04-01:
With the exception of joint damping (#14405), this is solved by #13860. I'll close it after I get #14405 in.

jwnimmer-tri, 2024-11-02:
> I'll close it after I get #14405 in.

It's in.

closed 2024-11-02 by jwnimmer-tri