RobotLocomotion/drake #11925: doc multibody: Describe preference of `p_BoQ_B` vs `p_BQ`?

EricCousineau-TRI, 2019-08-13:
Follow-up from this review comment:
https://github.com/robotlocomotion/drake/pull/11921#pullrequestreview-274565134

I haven't really been a fan of the overly verbose `p_BoQ_B`, which seems like it's fully described by the existing documented "terse" form, `p_BQ`, since `Bo` and `B` are both indicated by `B`.
I haven't dug in that deep, but I've briefly reviewed our docs for monogram notation, and don't see a call-out that states why `p_BoQ_B` is better:
https://drake.mit.edu/doxygen_cxx/group__multibody__quantities.html

Is there a brief explanation that merits this in all contexts, even in simple ones? If so, can that explanation be added to the docs?

I strongly prefer the terse notation, but can be swayed if there's a sufficiently simple explanation.
I can understand why it'd be useful in complex contexts where you want to call direct attention to differences, but it smells like extra noise in simple contexts.

Possibly relates #10798

sherm1, 2019-08-13:
Definitely not. IMO the terse notation is almost always better, unless there is some reason to emphasize the specific point or frame.

sherm1, 2019-08-13:
I think @mitiguy is the primary fan of the pedantic longer form. We allow both.

EricCousineau-TRI, 2019-08-13:
Gotcha. Then perhaps the resolution of this issue is just to encode that in the doxygen for multibody notation? (terse vs pedantic)

There is mention of equivalent monograms, e.g. `The corresponding monogram equivalents are p_BoBcm_B and p_BBcm, respectively.`, but no statement of when to prefer them.
Perhaps just adding the sentence, "Use of either short or expanded form is up to the author"?

EricCousineau-TRI, 2019-08-13:
@mitiguy I think my minor complaint gripe is the use of the phrase "Use standard notation" in the review, so perhaps if I had seen `Use standard forms of either p_BoQlist_B or p_BQlist`, it might've simplified this question in my mind?

mitiguy, 2019-08-14:
Yes - either is OK for this code base..
FYI: Although the terse one is fine, it confuses those folks who only need to use the notation occasionally.  I work with the automotive team to standardize notation, and some have problems digesting more than one form -- with implied concepts being more problematic.  Some have trouble distinguishing a point from a position vector and some struggle with Drake's p_BQ means p_BoQ, not p_BcmQ (herein B resolve to B's origin Bo, not B's center of mass Bcm).  I had asked Andy Ruina what he meant when a "body had a position" and he thought it meant Bcm.

mitiguy, 2019-08-14:
Yes - either terse of explicit notation is fine for the Drake code base..
I am going to close this issue -- please reopen this issue if appropriate.

Note: My comments in PR #11921 are focused on using "position" where a generic vector was in use and using a generic "v" where "position" is more appropriate.

closed 2019-08-14 by mitiguy