RobotLocomotion/drake #11931: Notation in ContactSurface should match camera ready version of hydroelastics paper

edrumwri, 2019-08-15:
In particular, we now need to rename E, which was the generic field for computing the contact surface, to `pknot` which is effectively the name used in the cited paper.

sherm1, 2019-08-15:
`pknot`? Did you mean `pnaught` to represent the symbol p₀ from the paper? If so, I would suggest `p0`  instead.

sherm1, 2020-02-25:
Damrong, I'm not sure if this is still an issue (it was assigned to Evan). Please close if not. Note low priority - not urgent.

DamrongGuoy, 2020-02-25:
Right now the public API of `ContactSurface` uses `e_MN` including `EvaluateE_MN(face, barycentric)`, `EvaluateE_MN(vertex)`, and `e_MN()`.  

We should change to `EvaluateP0()`, `p0()`.

DamrongGuoy, 2020-07-24:
We might want to remove the discussion of hₘₙ from the documentation of [ContactSurface](https://drake.mit.edu/doxygen_cxx/classdrake_1_1geometry_1_1_contact_surface.html) since ∇hₘₙ is not in the API anymore.

SeanCurtis-TRI, 2020-07-27:
Well, it's still worth discussing because we use the surface normals of the contact mesh as a piecewise-linear approximation of ∇hₘₙ.

jwnimmer-tri, 2024-03-20:
We seem to be OK without this.  If there is a 30 minute fix, anyone is still welcome to submit it, but it seems this work isn't sufficient important to schedule for the past several years.

closed 2024-03-20 by jwnimmer-tri