RobotLocomotion/drake #11936: Setting ImplicitIntegrator::set_reuse(false) should (or a new flag should be made to) trigger full Newton-Raphson

antequ-TRI, 2019-08-16:
With the current implicit integrator logic in drake, when set_reuse() is true, the implicit integrators attempt 3 "trial"s of increasingly complex Jacobian and iteration matrix refreshes. (Following the logic in Hairer 1996 book 2, pages 118-121)
When set_reuse() is set to false, it simply triggers the implicit integrators to calculate the jacobian once at the beginning of the timestep, and then attempt one trial without refreshing the jacobian during its iterations.

An example of where I'd like to see this changed is in StepAbstract() in implicit_euler_integrator-inl.h .

The full Newton-Raphson should perform a Jacobian recomputation every iteration during a trial.

Furthermore, under the full Newton-Raphson, the integrator should know that if fixed step is requested, it should not try to decrease the time-step size. The convergence-controlled time-step size reduction was proposed in Hairer 1996 (bk 2 pg 118-121) to accompany the quasi newton. Ideally, I'd like there to be a flag in ImplicitIntegrator (or IntegratorBase) that enables it to perform as a truly fixed integrator and throw during failure. (currently this logic is in IntegratorBase, IntegrateNoFurtherThanTime(), in integrator_base.h in these lines:
` if (this->get_fixed_step_mode()) {
    T adjusted_dt = dt;
    while (!Step(adjusted_dt)) {
      ++num_shrinkages_from_substep_failures_;
      ++num_substep_failures_;
      adjusted_dt *= subdivision_factor_;
      ValidateSmallerStepSize(dt, adjusted_dt);
      full_step = false;
    }
  } else {
    full_step = StepOnceErrorControlledAtMost(dt);
  }`
)
Thanks, and let me know if anything is unclear! This issue arose from our face to face discussion last week.

antequ-TRI, 2019-08-16:
tagging @amcastro-tri @sherm1  for reference

edrumwri, 2019-08-16:
Excellent summary of the conversation @antequ-TRI

edrumwri, 2019-11-20:
@antequ do you still believe this feature is necessary? Your experimentation after you filed this issue served to convince me that full Newton-Raphson is unnecessary (disabling reuse is only desirable for debugging, I think).

amcastro-tri, 2019-11-20:
IMO having the chance to enable full Newton-Raphson would help especially when debugging? otherwise we need to go through this logic every time we'd like to enable that (since most likely we forgot what's in there). I'm sure however @antequ developed stronger opinions now working with the integrators.

edrumwri, 2019-11-22:
@amcastro-tri Would you elaborate on the logic that we'd need to go through?

Also, would you describe what kind of debugging you'd be able to do with full NR that you can't do with almost-full NR? I assume that you imagine the debugging is useful for validating the reuse logic, but I think the almost-full-NR is sufficient to do so.

amcastro-tri, 2019-11-22:
IMO it is a very nice feature to have. Full Newton is easy to explain. Quasi-Newton with Jacobian computed only on the first iteration is also easy to explain. Hairer's logic definitely takes more words to explain. When debugging a particular situation, IMO going through less logic usually is a plus. Therefore that is why I am favor of supporting full Newton and quasi Newton with a single Jacobian updated at iteration k = 0.

sherm1, 2019-11-22:
I have also found full Newton nice to have around for answering nagging questions about why things are going wrong. Is it because I don't have a good enough Jacobian? Admittedly the answer is usually that that's not the problem, but being able to test with full Newton helps to clear away that hypothesis to make mental space for discovering the real answer.

edrumwri, 2019-11-22:
All right. I'm personally not convinced but I'll accept being outvoted. I've been resistant because I haven't yet thought of an elegant solution to incorporate full NR into the existing code.

antequ, 2019-12-05:
I don't have a strong opinion on this; just have two thoughts that people have already mentioned:

- Having full Newton-Raphson as an option will make debugging very easy for future integrator development
- It may be complex to add full Newton logic into the integrators, and it'll complicate the code in the different implementations of Step() even more. Furthermore, it's very difficult to write a test to ensure the flag is working. If we're willing to accept these drawbacks (and be flexible in how we test it -- the only test I can think of is trying a difficult ODE with a large timestep that works with full but not quasi newton), I think this is doable.

amcastro-tri, 2019-12-06:
> Furthermore, it's very difficult to write a test

The integrator can save statistics on how many NR are performed per step and how many of those iterations formed the full Jacobian (I believe we already do this?). Therefore IMO a good test would be that of a simple dynamical system for which when `set_full_newton(false)` those two numbers are different but we do observer the are the same when `set_full_newton(true)`. Presumably for a nice continuous dynamical system quasi-Newton works very well and we'd observe those two numbers do not match when `set_full_newton(false)`, and thus we should be able to make them match by using `set_full_newton(true)`.

edrumwri, 2019-12-06:
@amcastro-tri I think that's right- for a problem with a sufficiently long time interval, I expect there to be a difference in integrator stats (you're right, sufficient stats already exist for this) between full Newton, almost Newton, and quasi-Newton.

PR #12524 (2020-01-08): Adds ability to do full-Newton to implicit integrators.
Resolves #11936.

This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/12524)
closed 2020-01-08 by jwnimmer-tri