RobotLocomotion/drake #11937: direct feedthrough issue for pydrake

54WC, 2019-08-17:
Although I see some issue report for "direct feedthrough issue". But I am not sure how to fix this. I just convert a code from Jupyter version to python which running in Bash, then the issue happen.( Installation follow mit6.832 Text book Appendix A ) 
Do there is a convenient way to fix it ?  Thanks! 

The issue is described here:  
(1) The original code is: set_3.ipynb 
     http://underactuated.csail.mit.edu/Spring2019/set_3.html 
     I didn't suffer issue in this code. 

(2) set_3.ipynb is convert to inertial_wheel.py: 
    Please refer to the code here:
     https://github.com/54WC/direct-feedback-issue. 

(3) I clip and paste a piece code here :  
     def RunSimulation(pendulum_plant, control_law, x0=np.random.random((4, 1)), duration=30):
    pendulum_controller = PendulumController(control_law)
    # Create a simple block diagram containing the plant in feedback
    # with the controller.
    builder = DiagramBuilder()
    # The last pendulum plant we made is now owned by a deleted
    # system, so easiest path is for us to make a new one.
    plant = builder.AddSystem(InertialWheelPendulum(
        m1 = pendulum_plant.m1,
        l1 = pendulum_plant.l1, 
        m2 = pendulum_plant.m2, 
        l2 = pendulum_plant.l2, 
        r = pendulum_plant.r, 
        g = pendulum_plant.g, 
        input_max = pendulum_plant.input_max))
    controller = builder.AddSystem(pendulum_controller)
  
# Here is the problem
   builder.Connect(plant.get_output_port(0), controller.get_input_port(0))
    builder.Connect(controller.get_output_port(0), plant.get_input_port(0)) 

(4) error message: 

 RuntimeError: Reported algebraic loop detected in DiagramBuilder:
  InputPort[0] (u0) of System ::drake/pydrake/(anonymous)/Impl<double>/PyVectorSystem@00007f91c4061b10 (Impl<double>::PyVectorSystem) is direct-feedthrough to
  OutputPort[0] (y0) of System ::drake/pydrake/(anonymous)/Impl<double>/PyVectorSystem@00007f91c4061b10 (Impl<double>::PyVectorSystem) is connected to
  InputPort[0] (u0) of System ::drake/pydrake/(anonymous)/Impl<double>/PyVectorSystem@00007f91c4060e50 (Impl<double>::PyVectorSystem) is direct-feedthrough to
  OutputPort[0] (y0) of System ::drake/pydrake/(anonymous)/Impl<double>/PyVectorSystem@00007f91c4060e50 (Impl<double>::PyVectorSystem) is connected to
  InputPort[0] (u0) of System ::drake/pydrake/(anonymous)/Impl<double>/PyVectorSystem@00007f91c4061b10 (Impl<double>::PyVectorSystem)
A System may have conservatively reported that one of its output ports depends on an input port, making one of the 'is direct-feedthrough to' lines above spurious.  If that is the case, remove the spurious dependency per the Drake API documentation for declaring output ports. https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_leaf_system.html#DeclareLeafOutputPort_feedthrough

sherm1, 2019-08-17:
@54WC the link in the error message has an example showing how to specify state-only dependency for your output port -- did you try that? Our direct feedthrough algorithm has changed and might be reporting something that was handled differently before.

54WC, 2019-08-18:
I read Drake API document but not sure how to implement in this case in pyDrake.  Do you mean specify output port by "DeclareVectorOutputPort" or adding support for the symbolic:: Expression scalar type?
Would you please provide an example using the following code?   

def RunSimulation( ......)
       builder = DiagramBuilder()
       plant = builder.AddSystem(InertialWheelPendulum(.....)
       controller = builder.AddSystem(pendulum_controller)
       builder.Connect(plant.get_output_port(0), controller.get_input_port(0))
       builder.Connect(controller.get_output_port(0), plant.get_input_port(0))

sherm1, 2019-08-18:
I meant to change the DeclareVectorOutputPort() call in InertialWheelPendulum to specify "prerequisites" as is done in the link you provided above.

54WC, 2019-08-19:
I use DeclareVectorOutputPort() in InertialWheelPendulum as following, the direct feed-through error message is the same. 
Do I miss anything?  Sorry to bother you, I am quite fresh and don't understand well in this issue.  

class InertialWheelPendulum(LeafSystem):  
    def __init__(self, m1 = 1., l1 = 1., 
                       m2 = 2., l2 = 2., r = 1.0,
                       g = 10., input_max = 10.):
       LeafSystem.__init__(self)
       self.DeclareVectorInputPort("u",BasicVector(1))      # One inputs
       self.DeclareVectorOutputPort("x",BasicVector(4),self.CopyStateOut)   # Four outputs (full state) 
       self.DeclareContinuousState(4)

def CopyStateOut(self, context, output):
        if HasAnyDirectFeedthrough(self, input_port, output_port) != 1:
              x = context.get_continuous_state_vector(all_state_ticket()==0).CopyToVector()
              y= output.SetFromVector(x)

54WC, 2019-08-20:
It is work now by following code:  
self.DeclareVectorOutputPort("y",BasicVector(4),self.CopyStateOut,prerequisites_of_calc=set([self.all_state_ticket()]))

sherm1, 2019-08-20:
Great, yes that's what I meant @54WC.

54WC, 2019-08-20:
Thanks! 
The new issue happens, I don't know it is related to this modification or not.  I test another simple Pendulum case, the issue is the same. I might not gte everything right. 
 
Error message: 
Failure at bazel-out/darwin-py2-opt/bin/tools/install/libdrake/_virtual_includes/drake_shared_library/drake/systems/framework/system.h:1898 in DoCalcTimeDerivatives(): condition 'derivatives->size() == 0' failed.