RobotLocomotion/drake #11938: Python installation Mac - password

Jendker, 2019-08-18:
What is the reason to request the password by install_prereqs on MacOS? I am always careful when it comes to something like this.

jamiesnape, 2019-08-19:
Did you have Homebrew installed already? If not Homebrew requests your password to fix permissions in `/usr/local`. We do not directly request a password.

soonho-tri, 2019-08-19:
We ask not to run `install_prereqs.sh` under `sudo`. I guess while installing some of dependencies, `homebrew` needs admin privilege and therefore asks for it. (maybe java via bazel?)

jamiesnape, 2019-08-19:
Yes, also the `java` installed by `brew cask` since it calls an installer. That is unavoidable, I think, given the way JVMs work on macOS.

Jendker, 2019-08-19:
I see, thanks. I had Homebrew installed already so it had to be Java then.
I killed the `install_prereqs.sh` run and during the second time it did not ask me for password though, maybe I need to check if java was instaleled properly.

Issue solved, thank you for your time!
Btw. is there any chance, that drake will be available via pip or is it difficult to expect because of java and all other dependencies?

jamiesnape, 2019-08-19:
> is there any chance, that drake will be available via pip or is it difficult to expect because of java and all other dependencies?

https://github.com/RobotLocomotion/drake/issues/1183 is probably the best issue to follow regarding `pip`.

closed 2019-08-19 by jamiesnape