RobotLocomotion/drake #11939: Implicit Euler Integrator should initialize *xtplus in each trial

antequ-TRI, 2019-08-19:
Paraphrased from slack:
it turns out that a simple previous bug in the Implicit Euler integrator entirely explains why the convergence-controlled, fixed-step mode integrator used to choke for "simple gripper" with point contact; after this bug fix, it runs a lot more smoothly now with large timestep sizes. Here's my detailed explanation:
I noticed that in our Radau 1 implementation,  MaybeFreshenMatrices uses xt0 instead of the current iterate state, while in Implicit Euler, MaybeFreshenMatrices uses the most recent iterate *xtplus from the last trial. I commented on Evan's last PR ( https://github.com/RobotLocomotion/drake/pull/11754 ) about this difference, and Evan explained that the Radau 1 implementation makes more sense -- if the previous trial diverges, then it should start over using xt0 to refresh matrices.
It turns out, this was the bug in implicit euler that caused it to choke in the past.
By adding the line "*xtplus = xt0;"  right after this line linked here, implicit euler no longer chokes on "simple gripper":
https://github.com/RobotLocomotion/drake/blob/d0af1ec62bf5da72344800115adec0147251a9f0/systems/analysis/implicit_euler_integrator-inl.h#L145
Now under almost all "simple gripper" scenarios (both point and hydroelastic contact), the "convergence-controlled" fixed step integrator performs just as well with or without TALS (the iteration alpha limiter I'm working on). (That is, the time-step-size varying logic seems to be good enough)

Another minor change I'd recommend is changing this line:
`if (!this->MaybeFreshenMatrices(tf, *xtplus, h, trial,`
into this
`if (!this->MaybeFreshenMatrices(t0, *xtplus, h, trial,` or `if (!this->MaybeFreshenMatrices(t0, xt0, h, trial,`
(*xtplus and xt0 are equal at this point, but it makes it clearer that we are calculating the Jacobian at the beginning of the step and using only this Jacobian).

antequ-TRI, 2019-08-19:
tagging  @sherm1

antequ, 2019-08-23:
@amcastro-tri brought up a relevant note to this bug to consider for the future (but we should still fix the bug in the proposed way in the first post):

In the error-controlled integrators, after Implicit Euler is finished and succeeded, and there's a new *xtplus corresponding to x^{n+1}, that new *xtplus could be the starting iterate for Implicit Trapezoid.

I personally don't think it saves much performance-wise, but if we do want to implement that suggestion, we'd have to be very careful -- for example, if we ever let it take a Jacobian with this new *xtplus, and the Implicit Trapezoid fails, we would be in a very bad state --- the next step would incorrectly assume its Jacobian is fresh and evaluated at xt0, which can cause bugs. We must not change the assumption that the stored Jacobian was computed at xt0; we get choking or large errors if that assumption is broken.

edrumwri, 2019-09-07:
I just took a real look at this code. @antequ's analysis of the problem is right but I believe the solution he proposed is not ideal. The `StepAbstract()` function was designed to use a starting guess (as @antequ noted in his second post) using the rationale that the implicit Euler solution would be a good starting point for the implicit trapezoid iteration. Ante's solution would fix the problem he identified but force the implicit trapezoid to start from "scratch".

Rather than insert the line he recommended, the solution I am putting into place will set `*xtplus = xt0` _immediately before the recursive call to StepAbstract()`. This would have the effect of no longer attempting to iterate using a failed Newton iterate while preserving the ability to "warmstart" the implicit trapezoid method.

antequ, 2019-09-15:
@edrumwri that solution would still be very dangerous, in the one situation where IE succeeds and is very inaccurate and IT fails to converge in the first two trials, if it ever happens. It would initialize the Jacobian in the third trial with the inaccurate solution and potentially converge to the same inaccurate result, violating the error bound. The error bounds and quasi newton are all designed with the assumption that J0 is calculated from xt0. If you still want the computational advantage of having IT start from the IE solution, I would suggest keeping  your change but also in addition changing this line, 
> ```
>   if (!this->MaybeFreshenMatrices(tf, *xtplus, h, trial,
>       compute_and_factor_iteration_matrix, &iteration_matrix_)) {
> ```
to
> ```
>   if (!this->MaybeFreshenMatrices(t0, xt0, h, trial,
>       compute_and_factor_iteration_matrix, &iteration_matrix_)) {
> ```

I'll comment the same thing in the PR as well.

edrumwri, 2019-09-18:
My bug fix still allows doing that.

edrumwri, 2019-09-18:
@antequ I like that. To state another way, passing in xt0 would make it clear that that is what is already happening, while passing in xtplus elides that fact.

edrumwri, 2020-01-02:
Fixed the last remaining issue (passing in xt0 instead of *xtplus into MaybeFreshenMatrices()) in PR #12088.

closed 2020-01-02 by edrumwri