RobotLocomotion/drake #11944: Compute the kinematics chain for MultibodyPlant

hongkai-dai, 2019-08-19:
It would be great if we have a function in `MBP`, to return all the joints on the kinematics chains from one link to another link. This function is useful for imposing the sparsity structure in Jacobian matrix computation.

hongkai-dai, 2024-06-12:
Is it OK if we can support this functionality? We found that the solver performance can benefit greatly from constraint sparsity (see https://github.com/RobotLocomotion/drake/pull/21492#pullrequestreview-2086168608 with a factor of 6 improvement). Very often in inverse kinematics, the constraint only depend on a partial set of the joints.

sherm1, 2024-06-17:
FYI this is easily supported in the new MbP topology code. Let me see if I understand the request:
- we have arbitrary links R and S in a system with branches
- moving down towards World, there is a "highest" ancestor link A common to the paths from R->World and S->World
- you would like all the links (or dofs) in the R->A path and the S->A path, since only those dofs can affect a constraint acting between R and S

Is that correct? The MbP topology code already has an API for finding A but doesn't currently return the paths.