RobotLocomotion/drake #11951: ci: Provide weekly (or monthly) coverage of `no_{solver}` configs?

EricCousineau-TRI, 2019-08-22:
@jamiesnape did some work in #11945 to fix various bugs in `no_{solver}` configs, in response to #11942 for `no_dreal`.

Should we slate spinning up CI for this? Perhaps by mid-November?

@RussTedrake @jwnimmer-tri Can I ask y'all's opinion on this?

RussTedrake, 2019-08-23:
I’ll defer to you all on this one.

EricCousineau-TRI, 2019-08-23:
Sounds good.
My vote is yes, let's spin up CI by mid-November.

jamiesnape, 2021-03-12:
Can we remove the question mark in the title here (or close the issue if the answer is no)?

jwnimmer-tri, 2021-03-12:
I have not heard a reason why we need to officially support any no_{solver} config.  If someone has one, we can reopen this issue (or open a new one, if there's a more specific topic for support).

closed 2021-03-12 by jwnimmer-tri

jamiesnape, 2021-03-12:
Cool, an issue pruned. Apparently, I have almost 100 assigned to me across GitHub.