RobotLocomotion/drake #11954: pydrake doc sphinx: Constructors missing?

EricCousineau-TRI, 2019-08-22:
Was looking at `AxisAngle`, and saw that `__init__` docs weren't being shown:
https://drake.mit.edu/pydrake/pydrake.common.eigen_geometry.html#pydrake.common.eigen_geometry.AngleAxis_.AngleAxis_[float]

Same thing for stuff from `pydrake.math`:
https://drake.mit.edu/pydrake/pydrake.math.html#pydrake.math.RigidTransform_.RigidTransform_[float]

Going to try building locally.
Was this due to some upgrade? They used to show up...

EricCousineau-TRI, 2019-08-22:
Confirmed that building `master` (8a6e1bfbd) locally also does not show `__init__` strings.

Trying to build with `--config=python2`.

EricCousineau-TRI, 2019-08-29:
Jamie will check to see if it's due to my plugin hacks for templates.
To make names more friendly without the plugin hack, you can change this to just use underscores `x__T__U` rather than `x[T, U]`:
https://github.com/RobotLocomotion/drake/blob/a63e1e25afb0bc0ed80efc7faa1500dcfbfc0151/bindings/pydrake/common/cpp_template.py#L245

EricCousineau-TRI, 2019-09-19:
Per f2f, Jamie found that Mac is broken and Linux is only sometimes broken (Python 3).

May want to wait until we drop Python 2...

Jamie tried out removing the name mangling and disabling the extension, but it still didn't work on Linux Python 3.

For reference, here's the fix for pickling for (more hacks for) injecting stuff to be accessible via a module's `__dict__`:
https://github.com/RobotLocomotion/drake/pull/11976/files#diff-1b1ae8f6e3048fba92dee519ddd7f2deR318

EricCousineau-TRI, 2019-09-19:
@RussTedrake Can I ask if this is causing undue suffering with you + students? If it's mangeable, we may wait for us to drop Python 2 and then using Sphinx >=2.

RussTedrake, 2019-09-19:
I’m not suffering :-). It’s annoying, but i know enough to dig into the c++ and pybind to find the answers.

For class, i suspect students are fine until they start final projects in early Nov.

Fwiw - can you just upgrade sphinx now ?  Seems like python2 users not being able to build the docs locally is not really a concern?  And we want online docs to be geared to python3 anyhow?

EricCousineau-TRI, 2019-09-19:
Looking briefly, yeah, it looks like we could possibly disable `gen_sphinx.py` in Python 2, and then just use Python 3.

That being said, I actually was pointing some API stuff to @siyuanfeng-tri today, and realized that I couldn't show him the full API because the constructors were missing :P

Jamie, would you still be up for switching it over to Python 2 for now, if it takes ~15min, and then we start working on upgrading Sphinx for Python 3 only? (assuming we can just do a final switch and not have any intermediate PRs which breaks stuff?)

jamiesnape, 2019-09-20:
> Jamie, would you still be up for switching it over to Python 2 for now, if it takes ~15min, and then we start working on upgrading Sphinx for Python 3 only? (assuming we can just do a final switch and not have any intermediate PRs which breaks stuff?)

Done.

EricCousineau-TRI, 2019-09-20:
Sweet! Updated title to make it more specific.

EricCousineau-TRI, 2019-09-20:
(sorry, accidental close)

EricCousineau-TRI, 2019-10-31:
Given that we've removed support for Python 2, ctor docs are missing again. Jamie mentioned he will be able to resume this within the next week or so.

m-chaturvedi, 2019-12-05:
There seem to be multiple questions, here, I think:

#### Why is the python2 and python3 behavior different?
For some `__init__`s (as name) the `_skip_member` inside napolean extension returns `False` for python2 and `None` for python3:
https://github.com/sphinx-doc/sphinx/blob/v1.6.7/sphinx/ext/napoleon/__init__.py#L424-L451
This happens because when `obj = __init__`  and `what = "class"`,  the code is not able to get the class, i.e. the `cls` variable.
It is so because, for example, for class `pydrake.attic.multibody.collision.CollisionElement`  we have the `qualname = PyCapsule.__init__` and `obj.__module__ = pydrake.attic.multibody.collision`.

--- 

#### How to correct it?
 One simple 'fix' could be:
https://github.com/m-chaturvedi/drake/commit/e0a54c9ba123a04ba2eee4d2705248045e0a3933
Following the documentation: 
 > [ The member is excluded if a handler returns `True`. It is included if the handler returns `False`.](https://www.sphinx-doc.org/en/master/usage/extensions/autodoc.html#event-autodoc-skip-member).  

We note that it doesn't tell us the behavior for `None`.

But, this may not be the proper way to do it because we're still not fixing the behavioral difference.  I am trying to make an issue there.
PEP 3155: https://www.python.org/dev/peps/pep-3155/
Also, there are some bugs in the code above like the stray `skip` variable.

EricCousineau-TRI, 2019-12-05:
The quick fix sounds like an awesome one, we should def. do that first.

As far as the core fix, it seems really weird that the `__qualname__` is `PyCapsule.__init__` rather than `{cls}.__init__` (which is what the rest of the code requires). Sounds more like a `pybind` issue, where some metadata wasn't passed to the `py::cppfunction` to override the qualname -- it seems like it works properly for methods, though?

m-chaturvedi, 2019-12-06:
>  it seems like it works properly for methods, though?

Which methods?  If you mean other methods, that's because `keep` gets set to `True` here for many methods (not starting with `__` etc.):
https://github.com/sphinx-doc/sphinx/blob/v1.6.7/sphinx/ext/autodoc.py#L979

---

If you mean there are still some `__init__`s for which there is documentation, its because of cases like this: 
https://github.com/RobotLocomotion/drake/blob/master/bindings/pydrake/common/containers.py#L96
Doc: https://github.com/RobotLocomotion/drake/blob/master/bindings/pydrake/common/containers.py#L96
They get the cls from `cls = obj.__globals__[cls_path]`
(Maybe pybind should be setting this?)

---

If there's some confusion because of:
> Jamie will check to see if it's due to my plugin hacks for templates.

It seems to me that this happens for non templates as well.
For ex (new doc): https://drake.mit.edu/pydrake/pydrake.manipulation.planner.html
Old doc: https://ibb.co/v1Lhnp7

-----

Yeah, it might be a Pybind problem, I'm not a 100% sure, but I couldn't find any PEP that they offend yet.

EricCousineau-TRI, 2019-12-09:
Here's a min-repro that contrasts with / without the workaround:
https://github.com/EricCousineau-TRI/repro/blob/4512bdf58adb34d66458d6d6b5aa1ca10d5e489a/bug/pybind_sphinx_init/README.md

This only deals with `pybind11` (`upstream/master`), sphinx, and CMake.

Without workaround:
> [image]

With:
> [image]

closed 2019-12-09 by m-chaturvedi

m-chaturvedi, 2020-01-09:
Pybind issue: https://github.com/pybind/pybind11/issues/2059
Sphinx quesstion: https://groups.google.com/forum/#!topic/sphinx-users/ioD0eh8MJlU

PR #12452 (2019-12-09): 'Fix' sphinx to get __init__ documentation.
Resolves #11954

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/12452)