RobotLocomotion/drake #11956: SurfaceMesh Referring Triangles initialization is too expensive

antequ, 2019-08-23:
Currently SurfaceMesh stores the referring triangles list as a std vector of std sets (so that the triangles incident to each vertex is stored in a sorted order). However, upon its construction, it basically allocates a set on a heap for every vertex. This is relatively expensive, even for small meshes, because Hydroelastics quickly allocates new SurfaceMeshes for each contact query (CalcTimeDerivative) if the state changes. SurfaceMeshes is treated as a lightweight object constructed often in simulation.

Here is the data structure in question:
https://github.com/RobotLocomotion/drake/blob/441156a8e006d21d84c12db944815150c7bd72ee/geometry/proximity/surface_mesh.h#L359

Here's the expensive subroutine that takes a significant portion of hydroelastics runtime:
 ```
 void SetReferringTriangles() {
    referring_triangles_.resize(num_vertices());

    for (SurfaceFaceIndex i(0); i < num_faces(); ++i) {
      const int kNumVerticesPerFace = 3;
      for (int j = 0; j < kNumVerticesPerFace; ++j)
        referring_triangles_[element(i).vertex(j)].insert(i);
    }
  }
```
A few relevant notes from Slack:
Jeremy mentioned:
"FWIW std::vector + then std::sort is still probably faster then std::set"
I mentioned: 
"The elements are already inserted in order during construction (see for loop). (And std containers don’t support multithreading anyways. We can add a unit test to future-proof the sorted order guarantee, if anyone ever tries to multithread it.)" 
Sherm mentioned this (to reduce everything to one heap allocation):
"Here's a two-pass version that handles the arbitrary-count case:
- While building the contact surface, keep a count for each vertex of how many faces use it, and remember the total incidence count.
- Then, allocate an std::vector of length total_incidence_count.
- Run back through the vertices, providing each with the index where its incident faces are stored. Set the current count to 0.
- Run through the faces again, adding the face index to each of its vertices at their assigned location and increment the count.
That algorithm is space-optimal since the incidence vector is packed. I suspect that the time for the extra pass would not be noticeable since computing the faces is a lot harder than just visiting them and doing a few index operations."
Alternately, "preallocate an std::vector of length 7n", since the maximum number of faces that can share a vertex is limited to 7.

antequ, 2019-08-23:
adding @sherm1 @amcastro-tri @DamrongGuoy @SeanCurtis-TRI

sherm1, 2019-08-23:
That sounds like fun. Anyone want to implement the two-pass version or something better? I'll be happy to do it when I can to it which is ... um ... not sure.

SeanCurtis-TRI, 2019-08-23:
FTR The "preallocate an std:vector of length 7n" cannot possibly be valid. There is no actual limit in the number of faces incident to a vertex.

Simple counter example: create a surface mesh from a sphere with longitudinal and latitudinal lines. If you have 64 lines of longitude, they all converge at the poles. So, the vertices at the poles each have 64 incidental triangles.

sherm1, 2019-08-23:
I do not agree that lazy evaluation is the right approach. Some disadvantages:
- the problem presented by this inefficiency remains, it is just deferred; if we actually need this information we have to compute it;
- the code will require mutable access to logically-const data, always awkward;
- lazy eval depends on the idea that we usually won't need this info -- I don't see how we can know that at this point;
- if we do have to speed this up the lazy eval code will be throwaway;
- there are possible performance advantages to be had be filling in this information at the time we're building the mesh, when most of the data will already be in cache.

My preference would be first to use a reasonable algorithm for this computation and then see if we still have a problem.

SeanCurtis-TRI, 2019-08-23:
@sherm1's response contained rather more passion than I would've expected. In f2f, I'm convinced that he didn't misunderstand me; he is just philosophically firmly positioned against lazy evaluation. The suggestion in any guise would evoke the same ire.

My value system differs, so I respectfully disagree. But I'm insufficiently passionate to fight for it. 

So, the assumption that was clearly communicated to me is that we can compute incidence cheaply enough on the fly (with the right algorithm) that even downstream consumers of the `ContactSurface` who don't require this data won't mind paying for it.

SeanCurtis-TRI, 2019-08-23:
Alternative question:

What purpose does the incidence data serve? Are we even computing it for a good reason. The current debate is predicated on the assumption that the *presence* of the algorithm implies the *need* of the algorithm's answer.

I suspect this isn't a real need. My argument is predicated upon the fact that right now, the `ContactSurface` is a triangle soup and the number of triangles incident to any vertex is always 1. Whoever is requesting the `referring triangles` is apparently working with this *reality*. And if it can work in this reality, it doesn't have to cloak it in the illusion of something more topologically sophisticated. So, we're paying a high cost to reproduce the data that is already intrinsic in the original surface data.

sherm1, 2019-08-23:
Just spoke to @edrumwri and he says he is no longer using this information. So one possibility would be just to remove this code!

amcastro-tri, 2019-08-23:
> Right now, this function always returns a single face in the set for each vertex. 

ditto @SeanCurtis-TRI's comment. 

This information provides nothing useful for our triangle soups. I'd remove this code.

PR #12438 (2019-12-05): Removed referring triangles from SurfaceMesh.
Resolves #11956

I'm not deprecating this code as we're no longer using it anywhere in the Drake code base and I have good reason to believe that no one is using it in other codebases either.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/12438)
closed 2019-12-05 by jwnimmer-tri