RobotLocomotion/drake #12010: Release v0.10.0

jwnimmer-tri, 2019-09-04:
We intend to tag a relatively recent master a as v0.10.0 source release, ideally this week.

The platform reviewers have asked for a minimally short release notes to also accompany the release.

jwnimmer-tri, 2019-09-05:
Looks like cdccfa02b73373c77f977f2ed1b235517fe379b7 is a good commit, passing all continuous and nightlies (with one still running but happy so far).

jwnimmer-tri, 2019-09-05:
Notebook of items to consider improving upon for future releases:
- Should have release process handbook (easy for any developer to follow).
- Release notes links into our sphinx documentation could / will bitrot.  Maybe find a better way to link them, or inline, or snapshot docs, or etc.
- Would be nice to have a better "Getting started from binary" text to link to.

jwnimmer-tri, 2019-09-05:
https://github.com/RobotLocomotion/drake/releases/tag/v0.10.0

closed 2019-09-05 by jwnimmer-tri

EricCousineau-TRI, 2019-09-05:
> Release notes links into our sphinx documentation could / will bitrot [...]

Should we spend any time trying to archive older minor / major versions of documentation?

jwnimmer-tri, 2019-09-05:
Probably eventually we would want that, but I'd rather focus on other release-related improvements ahead of that.  My thought for all of the "notebook items to consider improving" is to carry them in mind as we do releases, and as they are shown to be recurrently painful every time, we then open issues to fix them.