RobotLocomotion/drake #12027: PidController python binding seems to have memory corruption

RussTedrake, 2019-09-10:
Add the following line:
```
        self.assertEqual(controller.num_input_ports(), 2)
```
to
https://github.com/RobotLocomotion/drake/blob/master/bindings/pydrake/systems/test/controllers_test.py#L221

results in
```
test_pid_controller
    self.assertEqual(controller.num_input_ports(), 2)
AssertionError: -751256949 != 2
```

RussTedrake, 2019-09-10:
fyi @mpetersen94

RussTedrake, 2019-09-10:
blocking #12009

EricCousineau-TRI, 2019-09-10:
Can reproduce here: https://github.com/RobotLocomotion/drake/commit/378191ec35bd9674bc9e4102819287eec5f300c0
Debugging...

EricCousineau-TRI, 2019-09-10:
Ah, multiple inheritance... Didn't realize that this interface had that:
https://github.com/RobotLocomotion/drake/blob/838a9155/systems/controllers/pid_controller.h#L46-L47

I'm pretty sure that's the problem. Will see what I can do.

PR #12030 (2019-09-11): pid_controller: Reorder inheritance order to workaround pydrake errors
py controllers: Minor touch-ups for testing

Resolves #12027

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/12030)
closed 2019-09-11 by sammy-tri