RobotLocomotion/drake #12055: pydrake: Clean up `bindings/pydrake/solvers`

EricCousineau-TRI, 2019-09-13:
For legacy reasons, `bindings/pydrake/solvers` breaks convention with the rest of `pydrake` (as does `autodiffutils_py` and `symbolic_py`), per the "Module Organization" bit here:
https://drake.mit.edu/doxygen_cxx/group__python__bindings.html

This issue should be resolved once everything is under `solvers_py.cc`, possibly with carving out the other bindings to their own `*.cc` files (as is done for `framework_py.cc`).
Then, all symbols are imported from `pydrkae.solv

\cc @hongkai-dai

sherm1, 2020-06-09:
:+1: for reorganizing the pybindings! It is difficult for us amateurs to find our way around there.

From [discussion](https://github.com/RobotLocomotion/drake/pull/13505#issuecomment-641553091) in #13505:
> > Could you PR your changes for the DisableCaching stuff against my branch?

> The only reason I didn't do that is I don't know how to find my way around in the bindings. DisableCaching is a method of ContextBase rather than Context. I was able to stumble around until I landed on bindings for Context, which is mysteriously located in framework_py_semantics.cc. I hadn't lucked onto ContextBase so I just added the DisableCaching binding to Context to run my experiment. (Obviously not good enough to check into Drake.)
> Is there a magic trick that would allow me to locate bindings for a particular class within pydrake? Searching for the class name produces many false positives to wade through. One thought I had was that there weren't currently any ContextBase bindings but I don't know how I could know that for sure.
> I'm sure I can answer these questions myself after thrashing around long enough. That seems like an unreasonable effort for tiny additions to pydrake though. Suggestions?

EricCousineau-TRI, 2020-06-09:
Thanks! My attempt to identify some possible solutions to some of the issues you encountered:

- Provide better guidance on how to find where a C++ symbol is bound
  - Initial step is just suggestions on how to Ctrl+F - e.g. search for `"ClassName"`, which usually is a quick find for me.
  - Better step is traceability back to Doxygen symbol (but that may be sketchy), and somehow link that back to binding source. (That's a bit of a stretch).

- Ensure `ContextBase` (and similar classes) are bound.
  - I only bound `Context<T>` before since I wanted to minimize API surface area to start. I don't think that's a useful objective now.

jwnimmer-tri, 2020-06-09:
> Ensure ContextBase (and similar classes) are bound.

Some of the inheritance-oriented programming within `systems/framework` might be hairy to add into pydrake, given the pure virtual stubs and etc. that would be necessary.  If the only complaint is developer clarity, then improving up the comment here (and maybe also the docstring) might be all that's necessary:
https://github.com/RobotLocomotion/drake/blob/f9cb91582bb188094515d8af927644a25c6d86e6/bindings/pydrake/systems/framework_py_semantics.cc#L174-L182

sherm1, 2020-06-09:
Would it make sense to match the file names and class names as we do in C++ Drake? `context_base_py.cc` would be really easy to find! Maybe that could even live next to context_base.cc?

EricCousineau-TRI, 2020-06-09:
I wouldn't recommend that...
@jwnimmer-tri I'm happy to explain, or defer to you. Will wait for ~10min.

jwnimmer-tri, 2020-06-09:
Using grep with literal double quotes, i.e., `rgrep '"LeafSystem"' .` has exactly one hit (and the correct one) in about 90% of the cases, and only 1-2 hits in 95% of cases.  That is really easy and does not require reworking our entire physical source file layout.  Fixing the ContextBase comment (and others) to match that pattern would cover almost everything.

sherm1, 2020-06-09:
Oh, that might be the magic incantation I'm looking for! I tried it on a few classes and it worked perfectly. `"ContextBase"` is an exception -- just my bad luck to need that one.

rpoyner-tri, 2020-06-12:
Interesting solution. Does it merit any sort of enforcement to maintain the property?

EricCousineau-TRI, 2020-06-12:
Not sure how automatic enforcement could happen, but yeah, ensuring that a symbol that is indirectly bound should be mentioned in this form would help.

To Jeremy's point, I do think that `ContextBase` itself should be bound; it's non-templated, and can help reduce symbol bloat. In terms of `SystemBase`, this relationship was actually useful for public-facing API, and I could forsee the same thing happening here.

jwnimmer-tri, 2023-01-28:
FYI #18683 completes the cleanup of the `pydrake.solvers` modules.  Along with #11802, I plan to work on improving other modules over the next months.

closed 2024-03-21 by jwnimmer-tri