RobotLocomotion/drake #12057: RgbdSensor documentation doesn't specify in which direction the camera is looking (in camera frame B)

RussTedrake, 2019-09-16:
(sorry if I somehow missed it).

But I wasn't able to figure out which axes was front, right, and up in the camera frame by looking at the docs.  https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1sensors_1_1_rgbd_sensor.html

By contrast, the RigidBodyTree version does specify it:
https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1sensors_1_1_rgbd_camera.html

EricCousineau-TRI, 2019-09-17:
We had discussed that a bit, and we ultimately described it in `CameraInfo`, as it was most relevant there:
https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1sensors_1_1_camera_info.html

Perhaps just a more obvious cross-reference, and/or copy-pasta, would suffice?

SeanCurtis-TRI, 2019-09-17:
I've also strongly advocated for a simpler API aping the classic OpenGL mechanism of providing camera position, target position, and up vector. I've always found that to be *very* easy to reason about.

RussTedrake, 2019-09-18:
The link to CameraInfo from the RgbdSensor is only on the line about intrinsics, right?  So I didn't follow it when looking for the extrinsic info (camera frame)?

PR #14127 (2021-03-15): sensors: add documentation about the camera frame
I found it difficult to find this basic information when I was looking before.

resolves #12057

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/14127)
closed 2021-03-15 by SeanCurtis-TRI