RobotLocomotion/drake #12058: Removing gravity force

zulfiz, 2019-09-16:
Hi,

I am trying to simulate my robot as if it is hanging in air with no gravity field acting on it. According to the documentation, the rigid body tree automatically adds the gravity field. Is there any way to turn the gravity off? Thank you!

sherm1, 2019-09-16:
@zulfiz is there a reason you have to use the unsupported RigidBodyTree from the attic rather than the supported MultibodyPlant?

/cc @amcastro-tri

amcastro-tri, 2019-09-16:
This is supported with `MultibodyPlant`. You'd do:
```c++
MultibodyPlant<double> plant = // load model.
plant.mutable_gravity_field().set_gravity_vector(Vector3d::Zero());
```

amcastro-tri, 2019-09-17:
I believe this resolves your problem @zulfiz. Feel free to re-open this issue if you believe it still needs resolution.

closed 2019-09-17 by amcastro-tri

zulfiz, 2019-09-17:
Hey! Is there any way to do a similar thing with Rigid body tree. Actually the robot we are using uses an old version of drake. So, all the code base uses Rigid body tree instead of MultibodyPlant. Let me know if there is a way around it. Also if we can't turn off gravity, is there a way to hang/pin the robot in air?

amcastro-tri, 2019-09-18:
Browsing the source it looks like you can do `my_rbt.a_grav.setZero()`.
Note that we no longer provide support for RBT. If you have an sdf/urdf file, you could load the same model in MultibodyPlant, lots of examples in `drake/examples/multibody`.

zulfiz, 2019-09-18:
Thank you for your help so far. I have one more question. The robot we are modelling is a bipedal walking robot with a free floating center of mass. Is there any way we can pin the robot in air. By pinning I mean that is there any way we can say the center of mass of the system is fixed at certain (x,y,z) position. Even though the gravity is acting on it, but the robot doesn't fall down as it's free floating center of mass is attached to some fixed point. Thank you!

amcastro-tri, 2019-09-18:
You mean the actual CoM? (which changes with configuration). That would require some external force, no? are you thinking of a physical attachment?

zulfiz, 2019-09-18:
So physically we have fixed the free floating mass of the robot (Cassie robot by Agility Robotics) on a stand. The legs are hanging in the air. We are doing this to test some low level controllers for each joint. So, is there any way we could fix the free floating mass of the robot in air at some height above the ground? Thank you!

sherm1, 2019-09-18:
Do you mean just to weld some point of the robot to the World with a 0-dof "fixed" (weld) joint? Or maybe leave it free to reorient with a spherical (ball) joint?

zulfiz, 2019-09-19:
@sherm1 Yeah! In simulation we want to weld the free floating mass of the robot to a certain point in the world with 0 dof. The legs of the robot will however be free to move around. Below is the picture of the scenario we are trying to model:
[image]

Thank you for your help!

amcastro-tri, 2019-09-19:
MultibodyPlant has `MultibodyPlant::WeldFrames()`.
Say you want to weld the body of the robot (say it's called "RobotBase") to the world.
Then you'd do something like:
```c++
plant.WeldBodies(plant.world_frame(), plant.GetFrameByName("RobotBase"));
```