RobotLocomotion/drake #12069: Implement Jacobian transpose operator

amcastro-tri, 2019-09-18:
Per discussion [here](https://reviewable.io/reviews/robotlocomotion/drake/12065#-Lp-fejT9wYgnlYc-2lX) in PR #12065, we'd most likely like to remove the API in `internal::MBT::CalcInverseDynamics()` taking an `ignore_velocities` flag. The most common case with `ignore_velocities = true` (besides the internal use to compute the mass matrix) is when a user needs also `vdot = 0` to compute the product `τ = Jᵀ⋅F`.

This operator is also significantly more efficient (O(n)) than explicitly building the system Jacobian `J` (O(n²)) and performing the multiplication.

Therefore probably a good chance to at least  introduce this API in terms of our internal inverse dynamics with `ignore_velocities = true` and `vdot = 0`.

cc'ing @siyuanfeng-tri, @jwnimmer-tri for pointing this out.

Early adoptions of Jacobian transpose operator include:
- Gravity generalized forces, #12065.
- Generalized contact forces caching (soon to be used by joint reaction forces API).
- Optimized TAMSI solver to use O(n) forward dynamics.

sherm1, 2019-10-08:
The need for this cropped up again in @edrumwri's #12162.

edrumwri, 2019-10-10:
Yep- we wanted to transform spatial contact forces to generalized contact forces for MBP's existing (generalized contact force) output port.