RobotLocomotion/drake #12075: mkdoc.py does not generate docs for operator==

soonho-tri, 2019-09-19:
We have the following logic in `mkdoc.py`:

https://github.com/RobotLocomotion/drake/blob/a3f985fa547060b547f997188c56ff036fe778c9/third_party/com_github_pybind_pybind11/mkdoc.py#L85-L93

https://github.com/RobotLocomotion/drake/blob/a3f985fa547060b547f997188c56ff036fe778c9/third_party/com_github_pybind_pybind11/mkdoc.py#L133-L135

`operator==` is skipped since `operator=` is in `SKIP_PARTIAL_NAMES` and the pattern is matched with `operator==`.

cc @EricCousineau-TRI 
cc @jamiesnape

jwnimmer-tri, 2019-09-19:
Can you give an example of an operator== that has non-trivial documentation?

We might want to solve this with #12016 where you have to opt-in to a name for it, instead of scraping thousands of trivial (and empty) docstrings.

soonho-tri, 2019-09-19:
> Can you give an example of an operator== that has non-trivial documentation?

We have a few cases where we introduce a `EqualTo` method only on Python side which is using C++'s `operator==`. 

https://github.com/RobotLocomotion/drake/blob/a3f985fa547060b547f997188c56ff036fe778c9/bindings/pydrake/symbolic_py.cc#L185-L186
 
I thought that it's natural to use the documentation of C++'s `operator==` but found this issue.

jwnimmer-tri, 2019-09-23:
@soonho-tri The `bool operator==(const Variables& vars1, const Variables& vars2);` on master does not have a docstring yet.  The easiest way for me to resolve this is to have code on master that reproduces the problem, possibly with a `TODO` in pydrake where the docstring is missing.  Then, I can work on a PR to resolve that TODO.  Fixing gedankenbugs in mkdoc is difficult.

soonho-tri, 2019-09-23:
Got it. I'll do that.

jwnimmer-tri, 2021-11-10:
Closing out as "backlog unlikely to reach an actionable state within a reasonable timeframe".

closed 2021-11-10 by jwnimmer-tri