RobotLocomotion/drake #12086: point_cloud: Should permit SetFrom / operator= to update fields

EricCousineau-TRI, 2019-09-20:
See discussion item here:
https://github.com/RobotLocomotion/drake/pull/12079#issuecomment-533641237

Basically, `PointClolud`'s contract does not permit the following (in C++ or Python):
```
value = AbstractValue.Make(PointCloud())
# New fields introduced, not permitted by PointCloud.SetFrom.
value.SetFrom(PointCloud(new_size=0, fields=kXYZs | kRGBs))
```

This can be fixed by relaxing the contract for `PointCloud` to be less strict, and thus allowing the "structure" of the point cloud to be mutated and allow the model values to be more generic.

(As an alternative, perhaps we more seriously consider adopting `open3d`, if the dependencies are small enough?)

EricCousineau-TRI, 2020-05-02:
FTR We've discussed using `open3d`, both in C++ and Python, in Anzu.

We can see how painful this is (if it's painful at all). Ideally, we can leverage the fact that they use `pybind11` as well, and thus have API interoperability with our bindings as well.

jwnimmer-tri, 2023-04-23:
@zachfang could you take a look into this issue, please?  I think possibly #18987 has fixed this already.

PR #19255 (2023-04-24): [bindings] Use PointCloud object directly for GetInputPort.FixValue
Close #12086

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/19255)
closed 2023-04-24 by jwnimmer-tri