RobotLocomotion/drake #12087: Add nevergrad as a dependency to Drake

hongkai-dai, 2019-09-20:
As we discussed today in the meeting.

hongkai-dai, 2019-09-23:
From slack discussion with Jeremy

> We will need to add nevergrad and all of its dependencies individually to drake/tools/workspace/{myfoo}/** where myfoo is nevergrad, cma, bayesian-optimization, etc. for all values which do not have packages like python3-scipy already in Ubuntu 18.04.

> To scope the work you'll need to compute the transitive closure of its dependencies, and one by one look to see if Ubuntu has a new enough version or not.  Possibly it's only 3-4 packages missing, or might be 30 for all we know.

jwnimmer-tri, 2019-09-23:
It's even more complicated than that, with respect to installed Drake.  This might be the first python _runtime_ dependency that comes from pip / wheel?  I'm not sure that we actually want to install those into prefix, since most users will want to use their own pip / virtualenv for it?

hongkai-dai, 2019-09-23:
Based on slack discussion with Jeremy, adding `nevergrad` as a dependency in Drake would require a lot of efforts. On the other hand, it is simple for Drake's users to install nevergrad themselves (through pip or virtual env). So we will not bring `nevergrad` to Drake officially.

closed 2019-09-23 by hongkai-dai

RussTedrake, 2019-09-23:
I appreciate the dependency concern.  But writing the solver interface for nevergrad to understand mathematical programs is nontrivial and might add a lot of value.

Perhaps a separate repo?  Eg
drake-nevergrad?

jwnimmer-tri, 2019-09-23:
I understood this ticket to mean "Do the build system work that is a prerequisite for someone writing nevergrad bindings for mathematical program, so that someone who is interested in writing those bindings is unblocked".

My observation to @hongkai-dai was that `pip install --user nevergrad` (or better, a virtualenv) is all that is needed to unblock such code, so doesn't need any Drake build work leading up to it, and so this issue is moot.

I did not understand this ticket to mean "Publish mathematical program bindings for nevergrad".  I think that's a fine feature request if you like, either by re-titling this issue or opening a new one.

hongkai-dai, 2019-09-24:
We can have a separate repo (as Russ suggested), or I can have a separate branch on Drake (which assumes that the user has done `pip install --user nevergrad`), and then write a wrapper to use nevergrad to solve `MathematicalProgram`, like what we already have in anzu.

RussTedrake, 2019-09-25:
@jwnimmer-tri -- just to make sure I understand (your last post confused me slightly).  Is adding `nevergrad` to the list of `pip` installs in `install_prereqs.sh` sufficient or no?

@hongkai-dai -- I think we'd like it somewhere that can have CI coverage.  We can do circleCI, etc in a separate repo if it cannot land in drake master -- i do think that is a pattern we should ultimately encourage.  I think a dev branch is probably not a good solution?

jwnimmer-tri, 2019-09-25:
> Is adding nevergrad to the list of pip installs in install_prereqs.sh sufficient or no?

On Ubuntu, we are not using `pip install` in Drake, so no -- it's not sufficient in terms of calling `nevergrad` from Drake master.  However, a developer themselves could run `pip install` on Ubuntu locally in order to bootstrap their branch.