RobotLocomotion/drake #12091: Dense Articulated Real-Time Tracking

RussTedrake, 2019-09-22:
https://link.springer.com/article/10.1007/s10514-015-9462-z

The Drake developers at TRI have an implementation of this algorithm in our internal repositories. In principle, we are happy to share it, but making it public takes time and we have not yet prioritized it. Please upvote it by adding a "thumbs up" reaction to this comment if you want us to know that you are keen.

yenchenlin, 2019-10-13:
Hello @RussTedrake , thanks a lot for mentioning this! Any follow-up?

RussTedrake, 2019-10-14:
Your interest quite clear (thanks @mariabauza)!  We started talking last week about exactly how to bring it in and are budgeting time for it in our project planning.  It's not going to wink instantly into existence, but do we aim to deliver.

closed 2024-11-01 by jwnimmer-tri