RobotLocomotion/drake #12107: Bus error / abort trap related to Eigen usage under Xcode 11

jamiesnape, 2019-09-26:
Appears to be in release configuration only. One such failure relates to this line: https://github.com/RobotLocomotion/drake/blob/7d672344d45d8badd94b5bc76a47b138532c7bba/math/continuous_algebraic_riccati_equation.cc#L24

CC #11758.

soonho-tri, 2019-09-26:
FYI:
 - I've checked that using '3.3.7' does *not* resolve the issue.
 - However, using the [`3.3` backport branch](https://bitbucket.org/eigen/eigen/branch/3.3) fixes the problem.  There are only four bug fixes between the release of 3.3.7 and the tip of 3.3 branch. It seems that http://eigen.tuxfamily.org/bz/show_bug.cgi?id=1741 is the culprit but I haven't checked it further.

jamiesnape, 2019-09-26:
Appears to be literally the most recent commit that fixes  `//math:continuous_algebraic_riccati_equation_test`: https://bitbucket.org/eigen/eigen/commits/a5c3d12011a0a8f4b94a44a8b1c5d9139965c812

EricCousineau-TRI, 2019-09-26:
\cc @scpeters (just in case this is related to Clang bugs encountered with releasing GMP for Homebrew...)

jamiesnape, 2019-09-26:
I think it is related to optimization flags that we see it now, not a bug in the compiler.

jamiesnape, 2019-09-26:
(Though I may have a different bug that potentially a libc++ issue. Still triaging that, though.)

soonho-tri, 2019-10-03:
A small example (triggering segmentation fault):

```c++
// main.cc
#include <iostream>

#include "Eigen/Dense"

int main() {
  Eigen::MatrixXd B(2, 1), R(1, 1);
  B << 0, 1;
  R << 3;
  Eigen::LLT<Eigen::MatrixXd> R_cholesky(R);
  const auto result = R_cholesky.solve(B.transpose())(0, 0);  // Segfault
  return 0;
}
```

```bash
clang++ --std=c++11 -O3 -fstack-check -I <eigen_dir> main.cc
```

```
$ clang++ --version
Apple clang version 11.0.0 (clang-1100.0.33.8)
Target: x86_64-apple-darwin18.7.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

PR #12133 (2019-10-03): Work around incompatibilities between Eigen <= 3.3.7 and Apple LLVM 11.0.0
Closes #12107 (I guess).

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/12133)
closed 2019-10-03 by jamiesnape