RobotLocomotion/drake #12113: Centroid-subdivision of triangles in ContactSurface in ConstructSurfaceMeshFromMeshHalfspaceIntersection and CalcZeroLevelSetInMeshDomain.

DamrongGuoy, 2019-09-27:
Related to PR #12112. The same problem could happen in ConstructSurfaceMeshFromMeshHalfspaceIntersection() (mesh_half_space_intersection.h) and CalcZeroLevelSetInMeshDomain() (contact_surface_from_level_set.h).
[image]
 As the ContactSurface moves, some intersection triangles (triangle intersects halfspace, or levelset intersects tetrahedron) could change to intersection quadrilateral. Currently we subdivide the intersection quadrilateral into four triangles by adding its centroid, but we do not subdivide the intersection triangle.  As a result, we can have one triangle (from intersection triangle) abruptly change to four triangles (from intersection quadrilateral) as the ContactSurface moves.

A simple solution is to always subdivide an intersection n-gon into (n+1) triangles. As ContactSurface moves, we will have three triangles (from one intersection triangle) smoothly change to four triangles (from one intersection quadrilateral).
[image]

Since this centroid-subdivision of a polygon is common in the three codes for ContactSurface creation, we should factor out this function and make it shared by the three codes.

DamrongGuoy, 2019-09-27:
CC: @SeanCurtis-TRI @sherm1 @edrumwri @amcastro-tri @ryanelandt 

@ryanelandt I confirm that even though Figure 5 in PFC paper shows 5-gon-to-4-gon transition, the same idea applies to 4-gon-to-3-gon transition too, right?
[image]

SeanCurtis-TRI, 2020-10-08:
Closing this issue.

 - mesh-half space intersection was resolved by #12935
 - I've just created issue #14183 for removing the now unused contact surface from level set functionality.

closed 2020-10-08 by SeanCurtis-TRI