RobotLocomotion/drake #12118: Create geometry frameId for bodies without geometry

huihuaTRI, 2019-09-30:
Currently, the multibody plant won't create geometry frameId if a body does not have geometry specified in the urdf. However, a body without geometry is often used in different situations, such as attach sensors at different places.

sherm1, 2019-09-30:
Here is the relevant code from multibody_plant.cc:474 ish ...
```c++
template <typename T>
geometry::GeometryId MultibodyPlant<T>::RegisterGeometry(
    const Body<T>& body, const math::RigidTransform<double>& X_BG,
    const geometry::Shape& shape,
    const std::string& name) {
  DRAKE_ASSERT(!is_finalized());
  DRAKE_ASSERT(geometry_source_is_registered());
  // If not already done, register a frame for this body.
  if (!body_has_registered_frame(body)) {
    FrameId frame_id = member_scene_graph().RegisterFrame(
        source_id_.value(),
        GeometryFrame(
            GetScopedName(*this, body.model_instance(), body.name()),
            body.model_instance()));
    body_index_to_frame_id_[body.index()] = frame_id;
    frame_id_to_body_index_[frame_id] = body.index();
  }

  // Register geometry in the body frame.
  std::unique_ptr<geometry::GeometryInstance> geometry_instance =
      std::make_unique<GeometryInstance>(X_BG.GetAsIsometry3(), shape.Clone(),
                                         name);
  GeometryId geometry_id = member_scene_graph().RegisterGeometry(
      source_id_.value(), body_index_to_frame_id_[body.index()],
      std::move(geometry_instance));
  geometry_id_to_body_index_[geometry_id] = body.index();
  return geometry_id;
}
```

That side-effect of registering the frame with SG and adding it to MBP's maps should be extracted and made into its own API so a user (in this case Huihua) can call it without having to add any geometry. RegisterGeometry could then also use that API when there is no FrameId yet.

amcastro-tri, 2019-10-02:
could you comment on the problem you are trying to solve @huihuaTRI? FrameId's are created to talk with SceneGraph so, if a body does not have geometry, there's no FrameId associated with it. That seems ok thus far.
Now, why is it that retrieving the geometry-less body say, by name, is not enough for you?

huihuaTRI, 2019-10-02:
To be more specific, in FL's urdf, the whole head is lumped into one link, so only one .obj file for the whole head. However, the head includes 2 fisheye camera and 4 realsense camera. Then in the urdf, there are several links defined for these sensors. However, there is no geometry for these links.  What I am trying to do is to attach a camera to the frame of one of the fisheye cameras link, which does not have geometry. Hence, there is no FrameId associated with this body(or link). SceneGraph would not allow me to attach a frame is the parent body does not have a geometry FrameId.

You are right. If you put a very tiny sphere geometry for this link in the urdf, the problem would go away. However, per f2f discussion with @sherm1, we think it's necessary to add a function that allows users to add geometry frame even though there is no geometry defined in the urdf.

This seems possible seeing the code Sherm posted above. Hope I explain myself clearly.

amcastro-tri, 2019-10-02:
Another solution would be that we **always** register a `FrameId` for **all** bodies in the model. Then we wouldn't even need a new API. Would that work?

sherm1, 2019-10-02:
:+1: Yeah, that might be a better solution.

sherm1, 2019-10-03:
I do think that would be a better user experience. It still makes sense to have a FrameId corresponding to a BodyFrame regardless of whether there is geometry attached to the body. A non-geometry-carrying body can still _observe_ geometry, and that requires a FrameId (e.g. to attach a camera or orient a ray).

Deprecating that API would make sense to avoid the wrong implication, but it would work correctly in any case.

SeanCurtis-TRI, 2019-10-04:
My belated two cents:

1. I'm generally in favor of simply registering all frames. It's worth noting that this includes *more* than Body frames. It should also support arbitrary frames added in (e.g., a frame rigidly affixed to a body frame).
    - This makes the "automatically add everything" strategy a bit more involved. It means that every way of adding a frame has to ultimately include invocation of frame registration.
    - Every frame registered is one more frame pose to calculate on the MBP side and one more frame calculation to process on the SceneGraph side. Not the end of the world, but if the frames aren't used, then it's wasted work.
2. Adding an additional API where the user can opt in for other frames explicitly would resolve both of those issues.

That said, register everything works for me.

huihuaTRI, 2019-12-13:
I just want to bump up the discussion. 
This could be a blocker to the effort of using one set of urdf files between the FL stack and the Drake sim.

closed 2020-02-11 by huihuaTRI