RobotLocomotion/drake #12119: Improve CalcJacobianSpatialVelocity() documentation to explain why there is not an offset p_AoAi_A

mitiguy, 2019-09-30:
As requested by Evan.  In documentation for:
a.  MultibodyPlant::CalcJacobianSpatialVelocity()
b.  MultibodyPlant::CalcJacobianTranslationVelocity()

There currently exists and offset vector for frame_B of
param[in] p_BoBi_B

However, there does not exist an offset vector for frame_A of
param[in] p_AoAo_A

This is because it is not needed.
The Jacobian measured in frame A is the same for all points fixed on A.
Explain this situation.  
Note:  This relates to the fact that
velocity of a point Q moving relative to a rigid frame A can be calculated as
the time-derivative in A of Q's position vector p from _any_ point fixed in A.
For example, p may be p_AoQ or p_AcmA or p_AanyQ (where Aany is any point of A).

amcastro-tri, 2019-10-04:
Thanks @mitiguy! this relates to #10664, you might want to make a pass to that one as well.

mitiguy, 2019-10-07:
This issue is closed per PR #12135.

closed 2019-10-07 by mitiguy